cleanflight/cleanflight-configurator #238: Servo page shows incorrect values and not able to save

lkaino, 2015-08-26:
The servo page in current Cleanflight configurator master is not compatible with the Cleanflight master.

The MSP_SERVO_CONF command expects 7 bytes of data (min, max, middle, rate). Cleanflight puts the whole servo config in the reply (min, max, middle, rate, angleAtMin, angleAtMax, forwardFromChannel, reversedSources).

Configurator uses a separate MSP_CHANNEL_FORWARDING command to read the forwardFromChannel parameters for servos. To me it seems Cleanflight does not even support this command.

The configurator should read the channel forward + other servo config from the reply to the MSP_SERVO_CONF command.

I did not check the writing part but I bet it has similar issues.

fftunes, 2015-08-26:
Noticed that too. Luckily for me it's only the roll-servo channel. I could still use my camera tilt servo, although not being able to save servo range is a bit sketchy.

sppnk, 2015-09-02:
As I use airplanes now I had to change back to 1.8.1, because configurator in 1.9.0 can setup the rates of ailerons or anything. Hope it could be fixed soon.

sppnk, 2015-09-03:
[image]

Hi, this weird values is what I get in tab servos page. I'm trying to use the last experimental digitalentity fixed-wing version. Anybody knows where to find info on "servo" CLI command? Does "servo" command work in 1.10 or have I to use "smix" command? 

The documentation says "smix" only can be used with CUSTOM mixes, so if I use AIRPLANE mix I am unable to configure my servos with CLI

MauroMombelli, 2015-10-29:
seems like the configurator use the new servo protocol i proposed, BUT cleanflight use a (better) variation of it.

i'm fixing it right now

MauroMombelli, 2015-10-29:
fixed in https://github.com/inkonova/cleanflight-configurator/tree/tilting-1.10, see msp.js and servo.js (sorry it is mixed with tilt-arm code, but that shouldn't be a big issue)

hydra, 2015-11-10:
Please check latest firmware against latest development configurator.  If there are any issues please update this.

lkaino, 2015-11-10:
Servo page is working on 0.66 configurator, thanks!

closed 2015-11-10 by lkaino