cleanflight/cleanflight-configurator #240: Sensor Tab seems to be not working

HO4X, 2015-08-31:
Sensors Tab causes Freeze of "communication between NAZE and computer" on Mac OSX(10.10.0) with NAZE32 on Cleanflight 1.9.0. After reconnecting, everything will work fine again, except Sensors Tab. Not sure if it is an bug in the GUI or in the NAZE Firmware.

thenickdude, 2015-08-31:
Couldn't reproduce with CF 1.10 development version or CF 1.9. Which sensor tickboxes do you have ticked? 

Do you have an OSD connected that is competing for attention? If so try disconnecting it.

HO4X, 2015-08-31:
So i tested it with the configurator version 0.63.2, on two different macs and serval connections (USB / Bluetooth). But it doesn't work, no difference between 0.63.2 and 0.65.0. I had only ACC checked, nothing else, but for me it seems to be freeze with the requests of sensor data.. 

My Setup:
- Quad X with SimonK ESCs and NAZE32Pro
- GPS/Mag/Gyro/Acc
- SoftwareSerial for Bluetooth(SOFTSERIAL1 @ 9600Baud)
- nothing else, no OSD etc.

Chrome Version:  46.0.2490.6 dev (64-bit) (newest) on both Macs.

I also tried it with the Naze32 Configurator App on my Android and the graph[acc] has worked well, i didnt tested the other sensors.. but it seems to be an bug of the mac version of the gui.

thenickdude, 2015-08-31:
I'm testing on a Mac too. The code is very much cross platform so there's
not likely to be any real difference based on OS.

Have you disabled your acc with "set acc_hardware = 1"?

If you enable developer mode on Chrome's extensions page, I think you
should be able to right click on the Configurator window and click" inspect
element". This should pop up the developer tools with a console that error
messages might appear on during the attempted sensor reads.

HO4X, 2015-08-31:
The ACC is enabled, after reflashing the naze, i kept everything as it is as default. 
[image]

thenickdude, 2015-08-31:
Looks like sensors.js:17 has a bug:

```js
SENSOR_DATA.sonar[i] = 0;
```

It should read:

```
SENSOR_DATA.sonar = 0;
```

I guess that Chrome 46 throws an exception but my Chrome 44 doesn't for whatever reason.

I'll submit a PR to fix it.

HO4X, 2015-08-31:
Yes thats it ! Merged it on my local Version. Thanks !

thenickdude, 2015-08-31:
Reopening until it's fixed for everyone else :)

Netsplits, 2015-10-09:
Just to confirm that it's also working for me too. Great!

hydra, 2015-10-14:
fix merged.

closed 2015-10-14 by hydra

samerhay, 2015-10-18:
I'm having the same problem, but i didn't understand how you fixed it. Could you please guys help me out? Thanks

Netsplits, 2015-10-18:
Fix is merged, just redownload and reinstall the configurator, or edit by hand the file sensors.js as thenickdude explains:

>Looks like sensors.js:17 has a bug:
>
>SENSOR_DATA.sonar[i] = 0;
>
>It should read:
>
>SENSOR_DATA.sonar = 0;
>
>I guess that Chrome 46 throws an exception but my Chrome 44 doesn't for whatever reason.

samerhay, 2015-10-18:
Thanks. at first I couldn't and didn't know how to debug cleanflight. but then i realized i can, if i download the source code and manually add it to google chrome. now it works, but i'm having tons of problems with CC3D, which I will post in a separate thread. thanks for your help.

eimiss, 2015-10-19:
yep, changing SENSOR_DATA.sonar[i] = 0; to SENSOR_DATA.sonar = 0; fixes it. But current version from app store is broken

samerhay, 2015-10-19:
I agree,  the current version still has sonar[i]

Donykurniawan, 2015-10-21:
hey guys, i have same problems here. my naze32 after i change parameters, and save and reboot, can't back ready status, can't do anythink ,only disconnect and conntect again . when i clik in sensor, all grafik not showing. and cant do anything.  can anybody show me how to fix this( step by step please), i new bee for quadracer. thank you.

Netsplits, 2015-10-21:
You have to redownload from here GitHub, it's de developer version and includes this fix. You'll need to install it using the Developer Mode in the Extensions option in Chrome.

Donykurniawan, 2015-10-22:
Thank you Netsplits, i use OS: Windows, Chrome: 46.0.2490.71, Configurator: 0.65.0. can you show me, where's i can download cleanflight from GitHub(working link). thank you

prokreat, 2015-10-22:
the sensor tab isnt working for me either on SP Racing acro board. i downloaded the ZIP and reinstalled into chrome via developers mode, but when i click on the sensor tab it doesnt show anything and then can no longer click other tabs and freezes cleanflight except i am able to disconnect.

Donykurniawan, 2015-10-22:
yup, i try, and not working too, still freezing, cant back ready status n all sensor grafik gone. i try to borrow my freand pc, and test my naze. all work fine. can back ready status and all grafik in sensor show. so this cleanflight problems.

grahame22, 2015-10-22:
Can anyone explain in plain English how to get the sensors tab working? Have tried installing older versions of Cleanflight but makes no difference

eimiss, 2015-10-22:
Easy: assuming you're on windows, search for "sensors.js" file in C:\Users\YOURSELF\AppData\Local\Google\Chrome\User Data\Default\Extensions
Inside that file, search for "SENSOR_DATA.sonar[i] = 0;" without the quotes. Change it into "SENSOR_DATA.sonar = 0;" — remove the "[i]"

eimiss, 2015-10-22:
>You have to redownload from here GitHub, it's the developer version and includes this fix. You'll need to install it using the Developer Mode in the Extensions option in Chrome.
>Or wait until the version in the store is updated.

Nope, the fix is not included.

prokreat, 2015-10-22:
thank you. definitely not in the github download... manually fixing it solved the issue.

grahame22, 2015-10-22:
Tried looking in ''C:\Users\YOURSELF\AppData\Local\Google\Chrome\User Data\Default\Extensions'' but there's nothing there relating to sensors or sonar? :/

grahame22, 2015-10-22:
Ok have found the sensors.js file, but am unable to open it? Do I need a particular program to do this?

grahame22, 2015-10-22:
Thankyou eimiss!! Figured it with your help and got it working :)

Donykurniawan, 2015-10-22:
Easy: assuming you're on windows, search for "sensors.js" file in C:\Users\YOURSELF\AppData\Local\Google\Chrome\User Data\Default\Extensions
Inside that file, search for "SENSOR_DATA.sonar[i] = 0;" without the quotes. Change it into "SENSOR_DATA.sonar = 0;" — remove the "[i]"

where is it. i cant find app data in windows explorer, c::\Users\YOURSELF\.....(appdata not showing)

thenickdude, 2015-10-22:
It is in the GitHub download, you just have to download the Development
branch:

https://github.com/cleanflight/cleanflight-configurator/tree/development

Hit the "download zip" button on the right side, and unpack it somewhere
useful. In Chrome, click Menu, More Tools, Extensions. Tick the "developer
mode" tickbox and click "load unpacked extension". Point it at the folder
you unpacked from the zip file. Done.

Donykurniawan, 2015-10-22:
i was do, thenickdude, but not working. still freezing

thenickdude, 2015-10-22:
What's the name of the zip file you downloaded?

Donykurniawan, 2015-10-23:
can anyone show me where sensor data please. when i search in C:\Users\YOURSELF\AppData\Local\Google\Chrome\User Data\Default\Extensions, i cant find sensor_data.

Donykurniawan, 2015-10-23:
i download cleanflight configurator master.zip

thenickdude, 2015-10-23:
Wrong one, please download the one from the link I gave.

Donykurniawan, 2015-10-23:
oww, when i open your link, same when i download in middle right . there is clone in destop and download.zip. if wrong where u mine
. thank you

thenickdude, 2015-10-23:
Yes, it's the same button as before, but this is the development branch
page rather than the "master" page which is the version on the Chrome Web
Store. The zip file is different and called
"cleanflight-configurator-development.zip".

Donykurniawan, 2015-10-23:
so, i must uninstall first and download again right

Donykurniawan, 2015-10-23:
yes, i found development zip. and extract. in crome\extension. why show Failed to load extension from: D:\RC\CleanFlight\cleanflight-configurator-development
Manifest file is missing or unreadable.

Donykurniawan, 2015-10-23:
i load unpacked extension, and browse my download cleanflight configurator-development, and Could not load manifest.

thenickdude, 2015-10-23:
Did you unzip it and go into that folder when it asks you to choose? It
works perfectly for me.

Donykurniawan, 2015-10-23:
yes, i unzip it. but Manifest file is missing or unreadable.

Donykurniawan, 2015-10-23:
ok. now i can show cleanflight  0.66.0. but still freezing . 
2015-10-23 @ 14:01:28 -- Running - OS: Windows, Chrome: 46.0.2490.71, Configurator: 0.66.0
2015-10-23 @ 14:02:41 -- Serial port successfully opened with ID: 1
2015-10-23 @ 14:02:41 -- MultiWii API version received - 1.13.0
2015-10-23 @ 14:02:41 -- Flight controller info, identifier: CLFL, version: 1.10.0
2015-10-23 @ 14:02:41 -- Running firmware released on: Oct 2 2015 14:57:31
2015-10-23 @ 14:02:41 -- Board: AFNA, version: 2
2015-10-23 @ 14:02:41 -- Unique device ID received - 0x669ff504853827067210932
2015-10-23 @ 14:03:13 -- EEPROM saved
2015-10-23 @ 14:03:13 -- Device - Rebooting

cant back ready status

Netsplits, 2015-10-23:
It hangs when always when rebooting from either tab? If you disconnect and reconnect works again? It works now the sensors tab?

Donykurniawan, 2015-10-24:
disconnect and conncet , it hang every i change setting, in every tap. sensor now work,( show grafik) but when i change baud rate, still freez(hang).