cleanflight/cleanflight-configurator #242: Tabs block when activating mmix & smix by CLI

sppnk, 2015-09-05:
In version 1.10, when saving smix and mmix mixes for airplane in the CLI (because  the servo tab doesn't work) all the tabs get blocked and unable to access them. By the bluetooth flexiport (CC3D atom uart3) is still possible to comunicate by the CLI.

sppnk, 2015-09-05:
If I send mixer CUSTOMAIRPLANE to the CLI (CC3D atom), the configurator hangs (I cannot open any tabs).

thenickdude, 2015-09-06:
If you change nothing on the CLI and simply do a "save", it takes a good 10 seconds after clicking a tab in order for it to change (the configurator must wait for it to reboot). I don't notice any difference in behaviour if I also add "mixer customairplane".

sppnk, 2015-09-06:
Thx thenickdude, May be my PCs fault, but I have tried with two different notebooks and the behaviour is the same. The version of CF I am testing is the last build of https://github.com/digitalentity/cleanflight
compiled from himself, for CC3D:
https://www.dropbox.com/sh/4r5374mos7471q2/AAAM4jGfPqAi9Lhm0KpT9Gaua/cleanflight_CC3D.navigation-rewrite.Sep06.bin?dl=0

The configurator connects well and shows info in the upper window, but I cannot open the tabs.

thenickdude, 2015-09-07:
Are you using the development version of the configurator? Development builds require development versions of the configurator to go along with them. Grab the development branch from here using the Download ZIP button on the right, and follow the "alternative way" installation instructions:

https://github.com/cleanflight/cleanflight-configurator/tree/development

sppnk, 2015-09-07:
I have just tried 0.65 (last build) and the problem remains. 

The program doesn't hang, so it connects well to de flight controller. But it goes to setup tab and it is impossible to open the other tabs. Shows cycletime 3502, but static numbers. It seems there are not communication, but there are not error on disconnecting or something in teh upper window. Still possible to connect trough OSD or bluetooth (CLI from EZGUI android apk).

thenickdude, 2015-09-07:
Note that there's currently a different bug that causes the sensor tab to
hang, that isn't the first tab you try is it?

sppnk, 2015-09-07:
No. If I try directly CLI tab or other tabs I have the same problem. Always the cycletime freezes. If I press "calibrate accelerometer" button nothing happens.

hydra, 2015-09-09:
0.66.0 is the latest development version.

closed 2015-09-09 by hydra

bradyjoh, 2015-09-26:
I had the same issues with the crashing with my Naze32 in 1.9.0 and my CC in 0.65. With the Naze updated to 1.10 RC2 and CC in 0.66 I was able to set the mmix settings without issue!

sppnk, 2015-09-26:
Thx, I will try