cleanflight/cleanflight-configurator #247: Suggestion - add the 3D throttle midpoint setting to the configuration tab

BlakeFPV, 2015-09-21:
Quick suggestion / request for the next version of the CleanFlight app - include the 3D throttle midpoint setting to the configuration tab under the other throttle settings - currently it is only available in the CLI. More pilots are trying out 3D now, and this is a critical setting to keep the motors from spinning when disarmed or when fail safe is triggered.

NightHawk32, 2015-12-29:
@BlakeFPV thaks for your feedback.
This was added with 31cd2de3d38e95439c56f932a51a8c54b0cf2335
[image]

closed 2015-12-29 by NightHawk32

BlakeFPV, 2015-12-29:
Nice! Thanks for considering this, and all the great work that you guys do!

Date: Tue, 29 Dec 2015 13:33:07 -0800