cleanflight/cleanflight-configurator #251: Servo Order REVERSED in Cleanflight Configurator [Motor] Tab

AustinSaintAubin, 2015-10-04:
Cleanflight Configurator displays servo outputs in reverse order on the [Motor tab](https://github.com/cleanflight/cleanflight-configurator/blob/master/tabs/motors.js).
Issue exists in Cleanflight Configurator (0.65.0) & (0.66.0 Development)
This MIGHT be an issue with the firmware and not with the configurator... not sure.
Tested with Cleanflight Configurator (0.65.0) & (0.66.0 Development), 1.9.0 & 1.10.0 on SP Racing F3 Flight Controller (Deluxe)

| Hardware Pins             | Servo tab (Servo Names)   | Motor tab (Servos)           | SMIX Servos     |
|---------------------------|---------------------------|------------------------------|-----------------|
| Mapping & Display Correct | Mapping & Display Correct | Mapping WRONG. is backwards. | Mapping Correct |
| 1                         | 0                         | 8                            | 0               |
| 2                         | 1                         | 7                            | 1               |
| 3                         | 2                         | 6                            | 2               |
| 4                         | 3                         | 5                            | 3               |
| 5                         | 4                         | 4                            | 4               |
| 6                         | 5                         | 3                            | 5               |
| 7                         | 6                         | 2                            | 6               |
| 8                         | 7                         | 1                            | 7               |

hydra, 2015-10-14:
I have noticed this too.  Will fix now.

hydra, 2015-10-14:
Fixed in 7e6cdd5d1ded03df0a690c66d67acce8f8a75bd7

commit 7e6cdd5d (2015-10-14): Fix the servo order and numbering on the 'motors' tab.

Fixes #251.
closed 2015-10-14 by hydra