cleanflight/cleanflight-configurator #254: CLI stalls when accessed via VCP port

enly1, 2015-10-10:
I have observed that accessing cleanflight CLI on on an F1 board using MSP on VCP it seems to stall and stutter when doing larger transfers - like running 'dump'.  I also noticed the port utilisation value flash up a high value before returning to 0. Too fast to read the actual 

value from the screen update.

The same board when accessing CLI over USART1 works flawlessly.
The same board when accessing CLI from terminal emulator (putty) works flawlessly.

So seems to be something to do with the way that the configurator is managing the serial port and/or some interaction between USB VCP drivers and chrome.  

I I have seen the same thing on multiple development boards as well as a virgin CC3D board , so doesn't look to be hardware specific. 

Anyone got any ideas what it might be / where to look.

Long term I don't think i'm going to use VCP on my custom boards, but throw in a CH340G on usart1 instead to simplify reflashing anyway - but maybe beneficial for CC3D users if fixed.

enly1, 2015-10-10:
Video of issue (as not supported by github as attachment)

https://youtu.be/Qx5KjfJ4ZIo

andymac4182, 2015-10-12:
Anyone know if it would help to replace the driver like this? https://github.com/cleanflight/cleanflight-configurator/pull/250

enly1, 2015-10-12:
The stalling ONLY happens in the chrome configurator and its not actually using USB code for communication with the device as it appears as a com port to chrome.  Given that com port works perfectly on every other terminal client without stalls, it seems a little strange fot it to be the actual driver

So, with my logical hat on, I can't see why it would help, but will give it a whirl to see if we can work around.

andymac4182, 2015-10-12:
Was just a thought. I know chrome has some issues with certain drivers that was all.

enly1, 2015-10-12:
Yeah, everything is worth a quick try - swapped the USB driver this morning, but things didn't work out before I had to head off to work - the board wasn't responding at all, but not sure if it was good before I started, so just need a bit of time tonight to recheck after swapping out drivers.  Will confirm if it helps later tonight.

PS - thanks for the suggestion, didn't mean to seem like I was dismissing it...

enly1, 2015-10-12:
hmm ... 
https://code.google.com/p/chromium/issues/detail?id=473993

Seems like we're not alone ...

tracernz, 2015-10-12:
I'm not convinced that's the issue. The cTn wrote the MSP lib commands will be retried if they fail so losing a few chars from one shouldn't be an issue. And I've written Chrome apps using the serial API for VCP devices before with no retry and never lost any data.

enly1, 2015-10-12:
i've not checked if I actually lost any data or just stalled - need to
check tonight along with the usb driver swapout.  Will also change the baud
rate as well to see if it impacts anything.  Just strange as all other apps
monitoring that serial don't stall.  And with the serial port usage, it
seems like its just buffering up stuff before the app gets to read it all.
All confusing, but sure when we know what it is, it will seem simple.

enly1, 2015-10-12:
okay - so not getting very far with swapping com ports and also issues with drivers - basically anything other than using serusb driver and I don't get a com port any more, just ST Virtual Com Port or something similar ( and nothing can use the com port ).

Testing with different baud rates and settings makes no difference.  issue appears to be with configurator only still - shows up in cli but also in sensor data reads when set at fast updates.  Connecting to same device via USB/Serial board results in no delays.

Could someone else check out a CC3D on their machine - just view sensor data at 10ms each for acc/gyro and see if it stalls or remains smooth ?

enly1, 2015-10-12:
PS - No data lost - just stalls and then fast buffer dump (judging by the increased serial port utilisation figures shown)

enly1, 2015-10-12:
Okay, so the problem has been identified

The issue is that when processing data from a VCP, the data comes in in packets of 1 or two bytes at a time which gets thrown through the serial read loop in the cli tab.  This loop appends the serial data into the UI control and then requests a window scroll to the bottom each time.

When serial data comes from a USB/Serial adapter, the packet sizes are around 32 bytes each time,so the UI code requests a scroll on the main window upto 32 times less frequently.

The stalls can be masked reducing the amount of time $('.tab-cli .window').scrollTop($('.tab-cli .window .wrapper').height()); is called from the TABS.cli.read function.  Swapping out the UI update call to be only when a CR/LF is detected, it smoothes it all out but obviously isn't a fix - but confirms the theory.

I assume someone who knows the configurator and jquery better than I should be able to knock up a proper fix without too much trouble.  You should be able to detect if the packets are all small and then handle updates differently to when they come in larger blocks etc.

tracernz, 2015-10-12:
Good work. :) Hopefully somebody can sort it out. I will take a look in a few days when I have some time if nobody else has.

enly1, 2015-10-13:
Just for additional info, the same issue applies to other tab serial handling as well - its very obvious on the sensors tab with stalls and repeated sensor data readings on when using higher update rates, but strangely doesn't seem to impact the motors tab.

enly1, 2015-10-13:
Now we know its related to the small packets / frequency causing some issues, we have the two ways of fixing it - workaround in the UI and/or fixing the vcp output which seems to be already being discussed https://github.com/cleanflight/cleanflight/pull/1143 and https://github.com/cleanflight/cleanflight/pull/886

hydra, 2015-10-14:
@enly1 @tracernz  great work!

hydra, 2015-11-28:
@tracernz did you have a look at this?

hydra, 2015-11-28:
Also, I have been testing the firmware PR's mentioned in the firmware for a while and will be merging them soon.

tracernz, 2015-11-29:
I hadn't sorry. Just put this together now: https://github.com/tracernz/cleanflight-configurator/commit/3baf0fb1b28603d752fa8967de1133443c79a983
It seems to improve the performance a bit. Possibly you'd like the move the global variable to somewhere else.

closed 2017-02-25 by hydra