cleanflight/cleanflight-configurator #255: Sensors tab is broken

skaman82, 2015-10-15:
I just noticed that the sensor tab is broken. Tested it with two different computers and two naze32 boards. When I connect the board and click on Sensors. It shows nothing. When I try clicking on other tabs it doesn't react. I can just klick disconnect or restart the app to get to the other tabs again. 
[image]

qfather, 2015-10-15:
Can't reproduce this with 1.10 firmware or 1.11 build 2236 on a Naze32.

HaukeRa, 2015-10-16:
Have the same issue with 1.11 and the master branch of the configurator, the development branch works fine

borisbstyle, 2015-10-16:
I have the same issue on 1.10 and even on 1.11.

borisbstyle, 2015-10-16:
And indeed it works on dev version

skaman82, 2015-10-16:
Since 1.10 is released an the dev version of the configurator is not its not a solution to use the dev version.

qfather, 2015-10-16:
It is a prerequisite to use the dev version.  The very first line in the release info of the firmware states this.

skaman82, 2015-10-16:
@qfather yes you are right. IMO there should never be a CF version released without the full compatible version of configurator.

nwenkel, 2015-10-17:
Have had the same issue with the developer version. After reinstalling 0.66 the sensors tab works again.

Rhyell, 2015-10-23:
The issue lies in the sonars.js file. 

Line 17 reads 
SENSOR_DATA.sonar[i] = 0;

It should read 
SENSOR_DATA.sonar = 0;

JohnOCFII, 2015-10-24:
> The issue lies in the sensors.js file in the tabs folder.

Thanks.  With all the other Chrome related issues popping up lately, I had assumed this was related to the version 46 release of Chrome.  I guess not.  I've now pulled down a fresh version of the development configurator (a shame it isn't released yet) and the problem is gone.

MauroMombelli, 2015-10-29:
sensors.js:17 Uncaught TypeError: Cannot assign to read only property '0' of 0

hydra, 2015-11-10:
Unable to reproduce in development branch, closing.  Open a new issue if there's still problems.

closed 2015-11-10 by hydra