cleanflight/cleanflight-configurator #256: Suggestion: new GUI for configurator. Interested?

skaman82, 2015-10-15:
Hi there! I' using cleanflight almost 6 months now (since I am in RC) and I love what you guys put together here but I'm seeing some things that could be improved. I am a graphic designer and work in a advertising agency. In last 10 years I've worked on a bunch of different web-projects an learned a lot about UI & UX Design.

From my point of view CF is great but is becoming more and more complicated to use as new features coming in. Some things are even just in the wrong place or not have correct labeling that a normal user understands without going through docs. It just looks like it was made by developers for developers but not for a normal folks like me. The lack of documentation is another problem.

I would like to develop a new interface for the GUI and would cautiously ask if anyone is interested in this before I start. Especially @hydra . It makes no sense to put a tremendous amount of work in this if no one is interested in implementing this. I have no clue about programming and need some help understanding all effects that each input field has on the configuration. From my work I also know that you can't make everybody happy and some of the community will most likely don't like my approach. But we can discuss things of course :)

My goal is not just make the GUI nice and shiny but also complete rethink its structure and functionality. The are talks about removing the CLI. I think this is a good thing, but the GUI needs some more fields for configuration that are not there yet (where can I put in a mag_align value when the CLI is gone?)

Here some examples of my thoughts on improving things:
• There should be just one way to backup the full configuration and all profiles with the backup button - not using CLI
• Profile select field seems out of place. It affects way more just the PIDs so why is this in the PID tab?
• PID Setup is very messy. Each controller uses different variables for different things but the fields are labeled the same on each controller. Good example for that are the LEVEL PIDs which are not realy PIDs. 
• There should be a kind of tooltip for each field with a short description what this value does and the used variable name (if CLI is still there). 
• There should be only hardware specific fields - so if no baro and mag are present there should be no ALT and MAG PID fields in the PID tab to make the interface cleaner. 

I can't put a time frame on this yet but I'm guessing this will take some months to accomplish. So if someone is interested to take the configurator one step further please let me now. I can provide PSD files to work with for final implementing.

scooper1, 2015-10-16:
I think this would be a good idea 
It would be nice to see a better way of seeing what option combinations can be used for each type and revision of board 
Getting rid of the cli asap and the semi hidden options it brings would be good  maybe a experimental tab to hold untested new options

skaman82, 2015-10-16:
This is a quick idea how the interface could look like
[image]

skaman82, 2015-10-17:
Reworked the first screen a bit and developed another version with alternative navigation. I hope you like it. I'm waiting for @hydra to comment on this. If there is no requirement for a better UI yet so be it.

[image]
[image]

jarno83, 2015-10-17:
Nice :)

skaman82, 2015-10-17:
Here a possible concept of the GPS Tab with integrated waypoint interface. Maybe this is interesting for @digitalentity who is working on Waypoints in his NAV Rewrite. 
[image]

enly1, 2015-10-17:
To me these look great - but one quick question - where did the icon packs etc come from - you been busy with editing tools or were these from an existing icon pack.  If the later, are the Creative Commons / Open Source ?

skaman82, 2015-10-17:
I made most icons myself. The other come from http://www.flaticon.com. I can redo them by myself later if necessary.

enly1, 2015-10-17:
All I can say is great work - love everything about the layouts.  If cleanflight choose not to use them, please release them for use with other projects - I'm sure people would greatly benefit (hint: me :-))

skaman82, 2015-10-17:
Thank you, we see what hydra thinks

digitalentity, 2015-10-17:
@skaman82 
Absolutely awesome! I love the design. Please also consider the full parameter list similar to [APM Planner](http://diydrones.com/forum/topics/arducopter-3-0-ready-for-wider-use?commentId=705844%3AComment%3A1303976). I believe CLI would at some point dropped in favor of something like this.

skaman82, 2015-10-18:
@digitalentity
Thank you. I'm not using APM so it would be helpful if you can clarify. What do you mean by "full parameter list"? Could you provide a screenshot of that area in APM? I'm actually searching for good concepts for adding waypoints, especially in mobile apps where you can setup a mission with few touch commands. I think this has to work easy without having to set a bunch parameters.

digitalentity, 2015-10-18:
@skaman82 
Here is a screenshot of full parameter tree in QGroundControl (APM planner looks similar)
[image]

I'm thinking of this as a generic way of accessing all possible parameters in similar way we do through CLI. There should be an MSP counterpart inside the firmware, but that shouldn't be a problem. If such interface will appear CLI can be removed.

skaman82, 2015-10-18:
@digitalentity
You mean all parameters inside NAV, right? Could you tell me what each parameter does? Is this necessary to integrate in each waypoint or it is a global setting? NAV_PARACHUTE_EN IMO doesn't sounds like a necessary parameter.

digitalentity, 2015-10-18:
I'm talking about all parameters inside the flight controller. Literally **everything** that could be read or set by CLI or MSP.
There was a short discussion here: https://github.com/cleanflight/cleanflight/issues/922#issuecomment-141944259

skaman82, 2015-10-18:
ok, i think now I get it. You are not talking about waypoints, right? You suggesting to add a new tab with a parameter browser like in your screenshot, right?

digitalentity, 2015-10-18:
Exactly! I'm talking about all accessible parameters. If you can do it in
configurator, I'll create a flight controller counterpart for this.

-- 
Sent from Gmail Mobile

skaman82, 2015-10-18:
All I'm doing is the design concept for The UI. I'm searching for someone who can actually integrate this design and functionality in the configurator app.

skaman82, 2015-10-18:
I can do a mockup of the parameter browser. Would it be possible to also show the min, max an default values for each parameter and maybee a short description like in cli docs?

digitalentity, 2015-10-18:
Min and max - no problem, default - I don't think so. Default is set in the config code. For some parameters there is no default value (i.e magzero).

Parameter name and description should be stored inside configurator. Firmware should expose only IDs to keep footprint small enough and the configurator will have a mapping table, i.e. parameter 0x01000045 = "PID_ROLL_P"

scooper1, 2015-10-18:
Showing default values is a good idea. for values without default a blank field is a good indication of no default

skaman82, 2015-10-18:
@digitalentity 
This is how it could look like. The default value could be stored in configurator also.
[image]

skaman82, 2015-10-18:
Here some thoughts on backup functionality
[image]

tracernz, 2015-10-18:
@digitalentity A perhaps more user friendly impelementation (enum type fields have a dropdown with all possible values, units are shown, drop-down trees, only objects relevant to the connected board are shown, filter to search fields etc.):
[image]

skaman82, 2015-10-18:
@tracernz IMO: dropdown menu would be not so much user friendly on PIDs where we have every large amount of steps or in case of large numbers like in max_angle_inclination. The search field is a great idea though!

digitalentity, 2015-10-19:
@skaman82 looks excelent to me! Defaults are firmware dependent, but I have thoughts on how to implement them.
Search field as a great idea, but dropdown menus are a bad idea in general. However for some specific settings it might be a good idea - settings like "used PID controller", integer settings with 0 and 1 range (essentially a boolean flag)

tracernz, 2015-10-19:
@skaman82 Of course the dropdown fields are only for enums as I said! The boxes for integrals or floats are just boxes. The dropdown tree is much nicer to find stuff rather than one big list of variables.

skaman82, 2015-10-19:
@tracernz 
Have reworked the tree layout a bit. Added: 
• more depth (if necessary) 
• filter at the top 
• dropdown

[image]

skaman82, 2015-10-19:
@digitalentity 
After I thought about the GPS tag for a while I decided not to mix up GPS and waypoint management (when it maybe comes in the future). GPS tab should only about monitoring and maybe debugging. Now I did a separate tab for Waypoints and added some global Mission parameters, do you think this makes sense? And if it does which parameters cold be actually helpful?

[image]

[image]

digitalentity, 2015-10-19:
@skaman82  Looks great! A separate mission screen is definitely a good idea. Another suggestion would be a mission load/mission save button, so one can design a mission at home, go to the flying field and upload the mission without having to select all the waypoints again. What do you think?

skaman82, 2015-10-19:
@digitalentity DONE :)
[image]

digitalentity, 2015-10-19:
Great! I'm so loving your design :+1: 
I wish I knew Java so I could write this myself!

skaman82, 2015-10-19:
Yeah, I'm really hoping that somebody can make this happen. I think it is not to difficult to apply the UI. Its basically just html/css with some jQuery effects. The functionality is a whole other thing...

skaman82, 2015-10-19:
Got a bit carried away with layouts for new exiting functions like parameter browser and waypoints, will now focus on tabs that are actually there :)

skaman82, 2015-10-20:
My first shot at the configurations tab. Running out of space in my layout. Doing this in full size later. Any feedback? 
[image]

skaman82, 2015-10-20:
The "Edit custom mixer" Button could show a modal window with functionality like this: http://www.iforce2d.net/mixercalc/

skaman82, 2015-10-20:
@ledvinap @Pierre-A @borisbstyle @ProDrone Hi guys! I see you constantly work your ass off by improving cleanflight and fixing issues. I'm very grateful that you are putting so high amount of your time in that. I don't know if you are watching the configurator branch or not. I would very appreciate if you could take a quick look at my posts here and tell me what you think. Is it possible to achieve my proposal? I think cleanflight and many users would benefit from it. I don't mean the tons of new functions in my layouts – just the change of the styling of the configurator UI as a start. The new functions can maybe follow later if they make sense. Thanks!

borisbstyle, 2015-10-20:
@skaman82 Why dont you make a PR of this configurator version? I think Cleanflight UI needs to be renewed :dancer:

skaman82, 2015-10-20:
@borisbstyle I'm just a designer and have no coding skills, so this is just happening in Photoshop at this state.

borisbstyle, 2015-10-20:
@skaman82 Why not try integrate it into configurator :D

skaman82, 2015-10-20:
I could make the basic CSS for all used elements.

skaman82, 2015-10-20:
After a have all tabs finished in photoshop I will upload the resource PSDs and start to create CSS templates for each tab or area of the interface as far I get with my basic skills. In some places I will have to use placeholder images. From there on someone has to take over and integrate this into configurator. I can supervise from visual point of view and test things.

skaman82, 2015-10-20:
Here the full configuration tab. Some thoughts on this:
• This is actually one of the messiest tabs (from a designer point of view) and it is hard to keep so much information clean. 
• Renamed large amount of field labels. IMO the description what a feature actually does is more important user information that the CLI name. Thats why I put the description first.
• Placed notes above the inputs to match direction of reading.
• Reworked the QuadX picture to match the color scheme.

[image]

ProDrone, 2015-10-20:
I am not a GUI programmer, but totally love the stuff you design. So much that i am tempted to go there...

skaman82, 2015-10-20:
@ProDrone thank you. More exiting stuff is coming. Would really love to see this getting out there for people to use.

skaman82, 2015-10-20:
Do the Ports tab next. Try to take "IO - Cleanup timer/pin allocation #357" in consideration as a additional feature. Still not certain how the logic could be on the GUI side. Ideas? @ledvinap

skaman82, 2015-10-22:
Moving the project here: https://github.com/skaman82/CF_GUI

tricopterY, 2015-10-23:
I love the concept, but getting all of that working in concert with the background code is a fair bit of effort.
The Configuration Window looks cluttered. You may consider sliding-down windows for each group/panel?

skaman82, 2015-10-23:
@tricopterY Thanks für the feedback! I now that fancy stuff like parameter browser or waypoints is far away yet. That was just for fun. :) The first step would be to implement the design in the actual state of configurator and to add some small features like global profile switch or in gps tab a google map that shows the current GPS location if there is a internet connection and we have a fix. I think that is not so difficult for skilled people. I started to dig in the code an change some things but I can only do the CSS. I came across some problems already with JS. Have no clue about JS and need some help: https://github.com/skaman82/cleanflight-configurator/issues/1

tricopterY, 2015-10-23:
It's great to see an enthusiastic discussion, but unless @hydra (the project owner) endorses the concept, you cannot advance beyond everyone just voicing an opinion. @hydra must have his own good reason(s), for not participating. No one wants to see another variant of something that works reasonably well, just for some L&F change. Please do not get disheartened as this is the very nature of Open Collaboration, and no one should feel getting voted down.:smiley:

enly1, 2015-10-23:
And in exactly the same way, forks are produced when people have different
views of what would be beneficial to the community - isn't that how
cleanflight came about in the first place?

So while @hydra is the maintainer and has done a great job of taking
cleanflight and the configurator forwards and maintaining a clean code base
with well managed merges, it should not stop people pushing ahead with
updates that they think will benefit the community.  If people speak and
prefer this configuration tool, then i'm sure it will get the support it
needs regardless of the back-end.  Obviously it would be much better all
round if everyone agreed and worked towards a common goal, but lets not
kill this because @hydra has not commented or given it his blessing.

So - @hydra, what's your view - as @tricopterY has said, you have been very
quiet on the topic and while we know the cleanflight firmware is where your
focus is, you've usually checked in by now.

skaman82, 2015-10-23:
@tricopterY Thanks for the advice. The layouts I made are just a concept open for discussion. IMO it would be nice having a configurator that looks better. The current configurator still looks basically like baseflight and reminds me visually more of an excel sheet than of a modern app in some places. Cleanflight is a very great software so why not have a great presentation for it?  We all love nice things, right? :).

For me this is a way for giving back to the project for the fun I had using cleanflight. All I am writing in this issue is my professional opinion as a graphic designer with experience of many web projects. Conception and design of user interfaces is what I get usually paid for and what I love to do. 

I'm also surprised that @hydra still don't comment on this. 

@hydra: I would be very grateful for your thoughts on this. If you don't like it - no problem, but it will be good to know.

tricopterY, 2015-10-23:
Open Collaboration encourages Forks - but we need the necessary traction, for it to succeed.
@hydra is the prime mover and doing the heavy lifts.:bicyclist:

ProDrone, 2015-10-23:
There are so many things that need attention from @hydra that it can take a while even if he likes it.
Proposals in the form form of a good pull request, that require a minimum effort to merge, make a good change to get accepted.

Don't make it too fancy or change too much in a first proposal. A good first step may be just a visual make-over without adding new things (which is very tempting). So that the existing code can be used.

skaman82, 2015-10-23:
That is exactly what I would like to do as a first step. But changing the design alone are still thousands lines of code to review. It requires rebuilding things in some places. It is not my interest to create a new fork of this. I simply couldn't maintain it "under the hood" by the lack of my programming skills. I am far out my comfort zone by even trying to write the CSS myself.

digitalentity, 2015-10-23:
@skaman82 
Whie @hydra stays silen about this, may I suggest another feature? It would be great to have a wizart to calibrate the receiver (ask user to center all sticks, move them to their maximum deflection to properly set up rxrange feature). For the general idea please have a look here https://librepilot.atlassian.net/wiki/display/LPDOC/RC+Transmitter+Wizard (video from 1:02)

skaman82, 2015-10-23:
@digitalentity very good idea! noted

enly1, 2015-10-23:
@skaman82 - wasn't suggesting a fork was the way forwards - really don't want that to happen.  

I was just pointing out that if the worst comes to the worst, its an option.  If we're saying that if @hydra doesn't like the colour scheme or layouts then the whole thing is dead just seemed the wrong answer.  Lots of people have commented saying how much they like it, 

I would love to help, but can't commit the time at the moment.  I'm producing an open source data logger for motorcycle trackday/race use which is going to use a variant of the configurator for its setup/config which was why I commented on making it available anyway (for selfish-ish reasons). This does however mean that at some point I will have a decent handle on all of the configurators ins and outs and the time to help rework this, but need to prioritise my own project first.  

I've found a few cosmetic bugs already which I will backport into the main configurator code but its low on my priority list.

JohnOCFII, 2015-10-23:
Really impressive!  One of the things that would be great to integrate into any new design is some sort of "tool tips" where if you hover over an area, a bubble would pop up describing what that field was used for.  (Someone had mentioned that tool tips them selves might not be best, as tablet device like iOS have no concept of 'hover' but you get the idea...)

It's funny that in the broader world, Web programmers seem more prevelant than lower-level C programmers, but in this area, it seems more people are comfortable/willing to hack on the firmware C code than to work on the configurator.  I know personally that's true.

skaman82, 2015-10-23:
@JohnOCFII You can have a look at my layout of the full configuration tab from 3 days ago. I thought of that already :). I've read all feature requests on github regarding configurator an try to consider them all in the concepts.

JohnOCFII, 2015-10-23:
> You can have a look at my layout of the full configuration tab from 3 days ago. I thought of that already :).

Opps - sorry - I was not exploding the images in the Github comments, and started typing without reviewing the details.  It looks very good!

[10 more replies]

closed 2015-11-08 by skaman82