cleanflight/cleanflight-configurator #258: Arming switch for 3D mode doesn't work properly while connected to Configurator

phantom-8, 2015-10-17:
While setting up a 3D quad and connecting to Cleanflight Configurator and selected the Motors tab, the disarm switch doesn't work properly.  Here is the procedures to reproduce the problem.

- Setup a 3D quad and change the 3D parameters accordingly under CLI.
- Enable an arming switch in Configurator.
- Goto Motors tab in Configurator and arm the motors.
- Now, disarm the motors.  All motors will go full throttle in reverse.  (Actually, the motors will spin at the value set by min_command while disarm.)

The disarm problem only occurs if the quad is connected to Cleanflight Configurator and under Motors tab.  The disarm switch will work normally if not connected to Configurator.

Problem can be reproduced on Cleanflight 1.9 with Configurator 0.65, and also Cleanflight 1.10 with Configurator 0.66.

CerealKiller1, 2015-11-10:
I had the same issue however it happened all the time, plugged into config or not. I did not however try going back to 1.9 / 0.65.

I am also running SN20A's with one-shot on, arming with a switch and disarm at any throttle checked.

hydra, 2015-12-18:
fixed in latest development branch, please test and report back.

NightHawk32, 2015-12-29:
Yes this was fixed with d183f19369afa31208a498cacdbaa344ec8ca027 and 09100c8ebaa46146a9c6b727f001e14542cdf3fc
Already tested this with my 3D copter.

closed 2015-12-29 by NightHawk32