cleanflight/cleanflight-configurator #261: Waiting for data after reboot on changing tabs

ProDrone, 2015-10-20:
It is impossible to change tabs after a reboot via the configurator tab __PORTS__ and a click on __Save and Reboot__. The tab window stays blank with the message __waiting for data__.

After a Disconnect / reconnect all is normal again.

ProDrone, 2015-10-20:
I am trying to find the cause of this...

CASE 1 - Hooked up an __Afromini__ which has still some older version and was able to reboot succesfully:
~~~
2015-10-20 @ 17:04:28 -- Running - OS: Windows, Chrome: 46.0.2490.71, Configurator: 0.66.0
2015-10-20 @ 17:04:33 -- Serial port successfully opened with ID: 1
2015-10-20 @ 17:04:33 -- MultiWii API version received - 1.13.0
2015-10-20 @ 17:04:33 -- Flight controller info, identifier: CLFL, version: 1.10.0
2015-10-20 @ 17:04:33 -- Running firmware released on: Sep 7 2015 16:48:53
2015-10-20 @ 17:04:34 -- Board: AFNA, version: 1
2015-10-20 @ 17:04:34 -- Unique device ID received - 0x66dff494849835087133342
2015-10-20 @ 17:04:40 -- EEPROM saved
2015-10-20 @ 17:04:40 -- Device - Rebooting
2015-10-20 @ 17:04:41 -- Device - Ready
~~~
All worked as it should with this combination.

---
CASE 2 - Latest (development) version of cleanflight using a __naze acro and full__ showed the problem:
~~~
2015-10-20 @ 17:22:51 -- Running - OS: Windows, Chrome: 46.0.2490.71, Configurator: 0.66.0
2015-10-20 @ 17:22:53 -- Serial port successfully opened with ID: 1
2015-10-20 @ 17:22:53 -- MultiWii API version received - 1.13.0
2015-10-20 @ 17:22:53 -- Flight controller info, identifier: CLFL, version: 1.11.0
2015-10-20 @ 17:22:53 -- Running firmware released on: Oct 20 2015 16:20:04
2015-10-20 @ 17:22:53 -- Board: AFNA, version: 2
2015-10-20 @ 17:22:53 -- Unique device ID received - 0x673ff565249755087073025
2015-10-20 @ 17:23:00 -- EEPROM saved
2015-10-20 @ 17:23:00 -- Device - Rebooting
~~~
The message `2015-10-20 @ 17:04:41 -- Device - Ready` that is there in the first log is missing and i have to disconnect / reconnect to restore normal operation:
~~~
2015-10-20 @ 17:23:04 -- Serial port successfully closed
2015-10-20 @ 17:23:06 -- Serial port successfully opened with ID: 2
2015-10-20 @ 17:23:06 -- MultiWii API version received - 1.13.0
2015-10-20 @ 17:23:06 -- Flight controller info, identifier: CLFL, version: 1.11.0
2015-10-20 @ 17:23:06 -- Running firmware released on: Oct 20 2015 16:20:04
2015-10-20 @ 17:23:06 -- Board: AFNA, version: 2
2015-10-20 @ 17:23:06 -- Unique device ID received - 0x673ff565249755087073025
~~~

---
CASE 3 - For the next test with a __naze__ i went back to configurator v0.63.0 combined with cleanflight v1.8.1  - __But this combination showed the same wrong behavior !__

---
CASE 4 - I went back to the latest versions (of case 2), this time connecting to the board over bluetooth (excluding the use of the windows driver for the CP210x USB to UART Bridge).

Surprise: Now __the same__ configurator / firmware versions that failed above __works fine !__

This is strange because in CASE 1 above i used an external CP210x USB to UART based device to connect to the Afromini. Using the same driver. A difference may be that for the Afromini i just use the TX/RX line of that interface.

I hope someone may add more to this, because it is quite annoying the way it is right now.

---
To be continued...

stronnag, 2015-10-20:
A number of commentators (rcg) have noted that this does not happen on Linux.

ProDrone, 2015-10-20:
@stronnag
Thanks, i noticed that too.
It looks like a windows specific problem, but the driver in device manager shows: 
[image]
That does not look like it's recently updated.
Still puzzled by this...

ProDrone, 2015-10-20:
Maybe a change in google chrome code then?
Any hint is welcome.

stronnag, 2015-10-20:
I have no clues, I'm running Arch which provides frequent chromium updates (also 46.0.2490.71) and that is OK. Perhaps a chrome update that just tickles the Windows VCP driver, given your tests. Can you test an older chrome?

sam-ward, 2015-10-21:
I have just noticed the same behavior yesterday running Configurator 0.66
on Chrome (latest stable) on Win 10, connecting to CF1.10.0.  I changed to
both Configurator 0.66 (from 0.65) and CF1.10 at the same time, so i'm not
sure which one caused the change in behavior.

I was starting to wonder if it was just my machine :-)  I'm not sure how to
trouble shoot this, but please let me know if there is anything I can do to
help.

ProDrone, 2015-10-21:
I did some more digging and attempted to device a solution.

It is due to the chrome serial driver receiving a break condition when the flight controller reboots. This is a known issue in the chrome serial API and can IMO not be solved in cleanflight nor the configurator. It must be solved by the chrome devs.

Issue here: https://code.google.com/p/chromium/issues/detail?id=428588

tracernz, 2015-10-21:
Disconnecting and then re-establishing the connection won't clear the break condition?

ProDrone, 2015-10-22:
@tracernz Yes it does, but that is a workaround. A break condition should not destroy an open connection (as it was before).

tracernz, 2015-10-22:
It's needed to work around the Windows USB driver for newer boards anyway so might not be a bad idea. ;)

JohnOCFII, 2015-10-22:
> It is due to the chrome serial driver receiving a break condition when the flight controller reboots. This is a known issue in the chrome serial API and can IMO not be solved in cleanflight nor the configurator. It must be solved by the chrome devs.

Thanks for that.  I'm having similar issues with Chrome on the Macintosh too, so it seems broader than just the Windows piece.  Glad to see its on the radar of the Chrome folks.

JohnOCFII, 2015-10-23:
> It is due to the chrome serial driver receiving a break condition when the flight controller reboots. This is a known issue in the chrome serial API and can IMO not be solved in cleanflight nor the configurator. It must be solved by the chrome devs.

The issue  you linked is nearly a year old.  If this problem started recently, I think it is more likely related to something else in the new Chrome 46 release.

https://en.wikipedia.org/wiki/Google_Chrome_release_history

ProDrone, 2015-10-23:
Maybe. I am tried to go around it (disconnecting and reconnecting) but as soon as a break condition comes in it is over (connection destroyed). The documentation says https://developer.chrome.com/apps/serial#method-setBreak, that is a new addition in chrome 45 (not a year old) it is the outgoing direction though. But you never know what gets ^%$@# by the change.

AndersHoglund, 2015-10-24:
I only see this problem with my (one and only) Naze32, three Flip32's and one CC3D seems OK.
Ref: http://www.rcgroups.com/forums/showpost.php?p=33021102&postcount=6669
Configurator 0.65 on Win10.
Lists above also suggests Naze32 (AFNA, version: 2)  beeing bad, while "AFNA, version: 1" seems OK. 
How about other platforms?

ProDrone, 2015-10-24:
Is there a different USB bridge chip & driver on the Flip32's? or something else different?

AndersHoglund, 2015-10-24:
No, from what I can tell the same,but I can't see the chip markings. Have to tear down and de-PlastiDip. Same Si CP210x driver at least. Connects to Win10 on the same Com port,same vid 10c4 same pid ea60 etc etc. I.e. Can find any diff....so far.

ProDrone, 2015-10-24:
No need. I found that the TX line is floating on reboot. This is interpreted by the driver as a break condition. Before (older chrome version) this was apparently ignored, but now it causes a hangup.

Mounting a 10K pull-up resistor from TX to +5V takes the problem away. The difference between the Naze32 and the Afromini seems to be in load on the TX line. I just dropped my configurator modifications and will look for a solution in Cleanflight firmware.

ProDrone, 2015-10-25:
Further conversations over here (https://github.com/cleanflight/cleanflight/issues/1433) please..

closed 2015-10-25 by ProDrone

JohnOCFII, 2015-10-25:
> Thanks for that. I'm having similar issues with Chrome on the Macintosh too, so it seems broader than just the Windows piece. Glad to see its on the radar of the Chrome folks.

And just to help narrow things, I am -not- seeing this on my Mac.  My issues were different (and resolved).

gunzrus, 2015-10-25:
I'm having the same issue, saves but won't reboot. This is true on all 3 of my Naze 32 boards . Also will not restore a saved configuration. Using Windows. I even deleted and reinstalled the CF APP. How do I get the latest configuration "66".  Thanks as you my can already see I am a newb I hope someone can help

ProDrone, 2015-10-26:
Problem should be fixed with pull request https://github.com/cleanflight/cleanflight/pull/1436
Further conversations over here (https://github.com/cleanflight/cleanflight/issues/1433) please..