cleanflight/cleanflight-configurator #262: Chrome Update - Configuration and Receiver Tab issues

pilotgeek, 2015-10-25:
I'm noticing some strange behavior after the latest chrome release.  (Chrome Update - The stable channel has been updated to 46.0.2490.80 for Windows, Mac, and Linux. This release contains a critical update to Adobe Flash Player (19.0.0.226).)

In CF, the configuration tab doesn't reboot when you click save and reboot.  Instead it saves and stalls on rebooting.  If I navigate to a different tab, it says waiting for data with the 'thinking' lines.  If I unplug to re-boot, the save looks like it worked when I connect next.

The more serious issue is that when using the Naze 1.10.0 firmware, the Throttle, Yaw, Pitch, and Roll controls don't show as functioning on the receiver tab.  If I erase the board and flash back to Naze 1.9.0 everything works except pitch using the same setup that was connected with 1.10.  Ive tested this extensively with multiple rx's on clean brand new models in the tx.  (Spectrum Rx's (610, 6210, 7610) on a DX9.  I've continuity tested the wires with a multimeter and swapped signal wires into the Rx to double check it's not a connection issue or a Tx output issue.  When I swap signal wires, the control resumes to the 'wrong' channel and the pitch (on 1.9.0) remains static.  On 1.10.0, the Aux channels work, but the control surface channels don't.  If I swap an Aux1with throttle, then the throttle stick moves Aux1 on the receiver tab.

Short vid of whats going on:
https://youtu.be/hlkhvaQFAro 

Can someone please confirm if this is a bug related to the Chrome update on last Thursday (10/22/15)  or if I have a unique situation here.

Thanks

JohnOCFII, 2015-10-25:
Duplicate of #261

pilotgeek, 2015-10-25:
RCGroups thread pertaining to this issue.  Another user is experiencing the same exact thing, so I'm not the only one. :)

http://www.rcgroups.com/forums/showthread.php?t=2528108#post33030742 

Another vid showing the differences between firmwares using the same hardware demonstrating the problem:
https://youtu.be/CWrqDm3_huU

pilotgeek, 2015-10-28:
Turns out this may have been largely due to a hardware issue.  One of the pads on the Rev5 board disconnected from the board.  Even though my wire continuity tests showed a positive connection, the actual pad on the board disconnected for some reason.  Several others have reported the AETR controls not working on 1.10.0 and .65.  so, there might still be a bug, but for now it's resolved for me by installing a new Naze32.

I've since replaced the board and flashed to 1.11 on .65 with the latest version of chrome and everything works except the reboot issue.  Looks like the re-boot issue will be solved with an upcoming release sometime soon.  

I'm going to close the issue as I'm in the air again and everything seems to be working for now.

closed 2015-10-28 by pilotgeek