cleanflight/cleanflight-configurator #265: Waiting for Data Error after save and reboot

Tabsy123, 2015-11-02:
After any tab that has save and reboot clicked or save in cli etc,  the configurator then says "Waiting for Data"  and never works again ,  I have to disconnect from the FC  and connect again,   I have tried latest 0.67 dev and normal configurator , also tried multiple versions of Cleanflight on the FC and tried my 2 differnt computers and also tried 3 differnt Naze32 boards,  all do this every single time.   Its a Pain in the arse when setting up a new board and you have to save and reboot the board often.

fbrcdx, 2015-11-02:
Exactly the same issue for me.

JohnOCFII, 2015-11-02:
The issue has been resolved in the master version of the code.  The fix is actually in the firmware code and not the configurator code (although the problem became apparent due to a Chrome change).

See the link for more details:  https://github.com/cleanflight/cleanflight/pull/1436

I don't know when the fix will be released.

closed 2015-11-10 by hydra