cleanflight/cleanflight-configurator #274: Firmware Flasher Tab: toggle switches issue

skaman82, 2015-11-12:
The new toggle switches are always shown as off on restart, even when the saved state is checked. This can cause trouble if the user cant see what is actually activated on default.

skaman82, 2015-11-15:
@wanderzell could you maybe have a look at this?

tricopterY, 2015-11-15:
I guess you have to use something like:
```
 (new Switchery(elem)).markAsSwitched();
```

wanderzell, 2015-11-16:
Ok I will take a look on this.

tricopterY, 2015-11-16:
Ok - this is truly a pain and on examination: 
```javascript
Switchery.prototype.create = function() {
  this.switcher = document.createElement('span');
  this.jack = document.createElement('small');
  this.switcher.appendChild(this.jack);
  this.switcher.className = this.options.className;
  this.events = events(this.switcher, this);

  return this.switcher;
};
```
Switchery hides the checkbox and replaces it with a span that creates the L&F.
But does not actually change the checked/unchecked state of the hidden checkbox, i.e setting the checked property does not affect it. I guess have to use setPosition(true).

wanderzell, 2015-11-16:
@tricopterY are you working on this?
I think I have solved this. And yes it involves attaching separate event handler to switchery to setPosition() manually.

skaman82, 2015-11-16:
Great news guys! Thanks a lot for looking into this! Just create a PR against current development branch.

tricopterY, 2015-11-17:
Sorry - I'm not working on this, just poking around. I tried the new code/PR, and it behaves as before. Unless, I am overlooking/missing something.

wanderzell, 2015-11-17:
@tricopterY "as before" meaning ... not fixed for you or fixed? It went too late yesterday and wasn't able to test it thoroughly - the other places with checkboxes mainly.

skaman82, 2015-11-17:
Just tested this. So far I see, this solves the problem. Don't see other bugs regarding this.

hydra, 2015-11-28:
Fix merged, thanks everyone.

closed 2015-11-28 by hydra