cleanflight/cleanflight-configurator #284: CleanFlight Crashing Chrome

Nivek-the-Relic, 2015-12-02:
A couple of weeks ago the cleanflight configurator was working fine, but that afternoon when I went to launch it, it kept crashing chrome. It was coming up with a note about latest version needing permissions(I didn't write it down). I removed & re-installed and now it just crashes straight away. Baseflight seems to be ok... I have tried removing chrome altogether and reinstalling, but no luck.....still just crashes chrome.
Help please!!!!!

jtp755, 2015-12-02:
@Nivek-the-Relic Are you on Windows/OS X/Linux?

jtp755, 2015-12-02:
@Nivek-the-Relic Check these out:
OS X: Use AppCleaner to remove Chrome and reinstall
Windows: http://www.howtogeek.com/140464/how-to-manually-uninstall-a-globally-installed-chrome-extension

Nivek-the-Relic, 2015-12-03:
@jtp755 I am running Windows 7 and I have tried removing Chrome & reinstalling, no change.

jtp755, 2015-12-03:
When you removed Chrome did you clean up the directories and such in your user profile? Thats what the article above is doing...expect its targeting the specific extension. This data does not get changed/cleaned up in you uninstall Chrome.

Nivek-the-Relic, 2015-12-04:
Ok, so I went in and deleted all of the extensions, deleted Cleanflight and uninstalled Chrome. Then restarted, downloaded chrome & reinstalled. I didn't log in to my Google account, downloaded cleanflight and it is still doing the same thing. Cleanflight is the only extension in chrome, except the standard google ones. Anything I could be doing wrong??

jtp755, 2015-12-04:
I cant think of anything you are doing wrong...that is definitely the correct steps.

What version of Chrome...just curious?

Nivek-the-Relic, 2015-12-04:
Version 47.0.2526.73 m

jtp755, 2015-12-04:
Hmm..im on Version 46.0.2490.86 (64-bit). Let me update and see.

jtp755, 2015-12-04:
no issues on my side with the latest version that matches yours. I am on OS X too.

jtp755, 2015-12-05:
At what point does Chrome crash? Are you able to load the Cleanflight app from the App Launcher or does it crash when connecting to quad?

Nivek-the-Relic, 2015-12-05:
As soon as I click the app, it crashes.

jtp755, 2015-12-06:
Any error message or anything? Screenshot?

Nivek-the-Relic, 2015-12-06:
When it was first doing it there was a message, as I put in my first post I didn't write it down, but it was something about permissions. I had to click on "confirm" (I think (KRAFT)) and then it would crash. That's when I tried to remove it and reinstall for the first time, after that it just crashes as soon as I click the app.

Nivek-the-Relic, 2015-12-06:
When I click the app, the configurator page comes up, then goes gray, the mouse pointer does the little waiting circle thing for a few seconds and then chrome drops and I get the “google chrome has stopped working” message.

jtp755, 2015-12-07:
Well dang man...

im outta ideas for now. Going to Google around and see what I can find.

tracernz, 2015-12-07:
Sounds like an upstream bug, it shouldn't even be possible for a javascript app to crash the browser.

jtp755, 2015-12-07:
Agreed but wouldn't that be reproducible to a degree? Maybe thats it though...i tested from OS X...not Win 7

jtp755, 2015-12-07:
@Nivek-the-Relic  Have you tried clearing all history and cache?

jtp755, 2015-12-07:
Also there is a button at the bottom of Advanced Settings in Chrome that will reset all settings. May be worth a try.

jtp755, 2015-12-07:
Not sure how far you are comfortable or want to take it but here are somethings that will help with enabling Chrome debug logging.

https://www.chromium.org/for-testers/enable-logging
https://code.google.com/p/sawbuck/

Nivek-the-Relic, 2015-12-07:
I have done CClean a couple of times along the way, no joy. A friend of mine suggested trying FireFox, do you think this may work? I will try your suggestions this afternoon and let you know. Oh and thanks heaps for the help so far, I would have no clue other wise!!!

skaman82, 2015-12-07:
Try this:
• Close Chrome
• Go to /Users/*yourusername*/library/Application Support/Google/ Folder and delete the Chrome folder. 
• Launch chrome again.

skaman82, 2015-12-07:
you also could try to delete com.google.Chrome.plist from /Users/*yourusername*/library/Preferences

skaman82, 2015-12-07:
@Nivek-the-Relic sorry, had this completely wrong, thought that you are on OSX

jtp755, 2015-12-07:
@Nivek-the-Relic Firefox will not help because the configurator is a Chrome App that runs in Chrome

Nivek-the-Relic, 2015-12-12:
Ok, so I have a debug log
[log.txt](https://github.com/cleanflight/cleanflight-configurator/files/59994/log.txt)
I have no clue what to do with this, so have fun with that!!

Nivek-the-Relic, 2015-12-27:
@jtp755 I know it's holiday season, but any chance you can discover anything in the log??
Or is there something else I can try? 
Thanks for the help !!!!!!

jtp755, 2015-12-27:
Sorry...totally missed your last reply. I'm out and about right now but
will take a look when I get home.

jtp755, 2015-12-28:
@Nivek-the-Relic It looks like you may have MalwareBytes installed. Can you disable that (and if possible any services from it that are running) and try again?

Also...are you a local administrator on the computer? Something else to try is to right-click and run Chrome as Administrator and then launch CleanFlight.

Also...how are you launching the app? Are you using the Chrome App Launcher or going to the web store and launching from there? Any difference between them?

Nivek-the-Relic, 2015-12-28:
I had MalwareBytes installed as part of my trouble shooting the other day, made no difference and have un-installed since. I have tried running Chrome as administrator this morning and no joy, from app launcher or web store. No difference between them. :/

jtp755, 2015-12-28:
Hmmm....let me think some more. I didnt see anything in the log that would explain it.

jtp755, 2015-12-28:
Do you get anything if you go to chrome://extensions in your address bar and then enable Developer Mode in the upper right-hand corner and then click Update Extensions now?

jtp755, 2015-12-28:
Try this:
Open Chrome. 
Go to chrome://extensions in your address bar.
Click on Details for CleanFlight-Configurator
Click on Create Shortcut and then Ok to create one on your Desktop
Right-click on the shortcut on your desktop and select Run as Administrator

jtp755, 2015-12-28:
Also curious how you installed Chrome. Did you install it the typical way (in your profile) or use the alternate installer to install for all users?

jtp755, 2015-12-28:
Ive tried every way i can think of to break it on a Win7 with latest Chrome VM and its worked every time in every configuration.

Are you running any virus scanners, intrusion detection apps, adware monitors, etc?

Run this tool: https://www.google.com/chrome/cleanup-tool/
Follow this to remove, cleanup Chrome install folders: http://www.wintips.org/how-to-completely-uninstall-re-install-google-chrome/

I know you have done the uninstall/reinstall before but did you remove all of the folders as well? CClean will not help with this or extensions in most cases.

Let me know the results and we can see what else to try. While I am inundating you with questions...if you dont mind...where are you located or atleast what timezone? If the above does not work maybe we can set up a time to do a join.me or WebEx so that I can see your screen.

Nivek-the-Relic, 2015-12-28:
I am east coast Australia, Queensland. A join up would be no problem.

Nivek-the-Relic, 2015-12-28:
I use Trend Micro, but that's it normally. I have followed the links you have suggested & no joy.

Arakon, 2016-01-03:
I see this same issue, except for me, the app starts, I can change things etc, then suddenly it simply closes without any error.
Only happened 3 times so far. Latest chrome, latest configurator, win 7 64bit. AVG Free virus scanner, but it has the browser extensions disabled.
This only started with 1.1.0, it never crashed on 1.0 before.

tracernz, 2016-01-21:
I also got this on OS X. The app window opens, stays blank/white for a few seconds and then Chrome crashes. It seems to have gone away after a couple of attempts to launch though.. Dump from a crash:
```
Process:               Google Chrome [59920]
Path:                  /Applications/Google Chrome.app/Contents/MacOS/Google Chrome
Identifier:            com.google.Chrome
Version:               48.0.2564.82 (2564.82)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Google Chrome [59920]
User ID:               501

Date/Time:             2016-01-22 05:37:16.691 +1300
OS Version:            Mac OS X 10.11.2 (15C50)
Report Version:        11
Anonymous UUID:        C36874EE-1114-2E21-232E-6D90EC2489A8

Sleep/Wake UUID:       A59A9F48-EFC7-4CEE-8713-160B0DEA8807

Time Awake Since Boot: 100000 seconds
Time Since Wake:       10000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   libdispatch.dylib             	0x00007fff836c157e _dispatch_alloc_continuation_free + 83
1   libdispatch.dylib             	0x00007fff836cec6b _dispatch_main_queue_callback_4CF + 1763
2   com.apple.CoreFoundation      	0x00007fff96ae6cd9 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
3   com.apple.CoreFoundation      	0x00007fff96aa1d3d __CFRunLoopRun + 1949
4   com.apple.CoreFoundation      	0x00007fff96aa1338 CFRunLoopRunSpecific + 296
5   com.apple.HIToolbox           	0x00007fff9463b935 RunCurrentEventLoopInMode + 235
6   com.apple.HIToolbox           	0x00007fff9463b76f ReceiveNextEventCommon + 432
7   com.apple.HIToolbox           	0x00007fff9463b5af _BlockUntilNextEventMatchingListInModeWithFilter + 71
8   com.apple.AppKit              	0x00007fff928c90ee _DPSNextEvent + 1067
9   com.apple.AppKit              	0x00007fff92c95943 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
10  com.apple.AppKit              	0x00007fff928befc8 -[NSApplication run] + 682
11  com.google.Chrome.framework   	0x0000000104b4bb6e 0x1045cb000 + 5770094
12  com.google.Chrome.framework   	0x0000000104b4b1f4 0x1045cb000 + 5767668
13  com.google.Chrome.framework   	0x0000000104b891a3 0x1045cb000 + 6021539
14  com.google.Chrome.framework   	0x000000010466bd05 0x1045cb000 + 658693
15  com.google.Chrome.framework   	0x0000000107a72cb7 0x1045cb000 + 55213239
16  com.google.Chrome.framework   	0x0000000107a75532 0x1045cb000 + 55223602
17  com.google.Chrome.framework   	0x0000000107a6e79d 0x1045cb000 + 55195549
18  com.google.Chrome.framework   	0x0000000104b06c8c 0x1045cb000 + 5487756
19  com.google.Chrome.framework   	0x0000000104b061f6 0x1045cb000 + 5485046
20  com.google.Chrome.framework   	0x00000001045cd772 ChromeMain + 66
21  com.google.Chrome             	0x0000000104364d42 main + 530
22  com.google.Chrome             	0x0000000104364b24 0x104364000 + 2852

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff91e9d6de __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f04a729 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff8f048365 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff91e9dff6 kevent_qos + 10
1   libdispatch.dylib             	0x00007fff836c1099 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff836c0d01 _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff91e9d6de __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f04a729 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff8f048365 start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff91e9d6de __workq_kernreturn + 10
[... 782 lines cut ...]
TOTAL, minus reserved VM space     1.0G      997 

Model: MacBookPro10,1, BootROM MBP101.00EE.B0A, 4 processors, Intel Core i7, 2.3 GHz, 8 GB, SMC 2.3f36
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe, 1024 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333531533642465238432D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333531533642465238432D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xEF), Broadcom BCM43xx 1.0 (7.21.94.136.1a1)
Bluetooth: Version 4.4.3f4 16616, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM256E, 251 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
USB Device: STM32  BOOTLOADER
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4
```
Quite clearly a Chrome bug since it's the main thread, and no way a Javascript app can cause a segfault.

closed 2017-02-25 by hydra