cleanflight/cleanflight-configurator #285: CLI becomes unusable after 'save' initiates reboot on Naze32 FCs

jtp755, 2015-12-02:
@CradyCopter and I confirmed that the CLI becomes unusable after executing 'save' with Naze32 FCs. 

Changes are saved and the FC is rebooted but the CLI no longer responds to subsequent commands. You have to change to a different tab and then back to enter more commands if needed. Screen shows 'REBOOTINGp'

I tested with Configurator 1.1.0 and Cleanflight 1.10 and Betaflight 2.1.4

CradyCopter, 2015-12-02:
Thanks jtp755!  Glad to know it wasn't my PC ; )

hydra, 2015-12-16:
it's because 'SAVE' does a reboot.  Do you have an idea of how the GUI should detect the fact that you're making the FC reboot?  What do you think should happen when it detects it?

hydra, 2015-12-16:
Adding some code to make it disconnect when 'Rebooting' is received via the CLI.

ledvinap, 2015-12-16:
@hydra: Just add `save` (and  possibly `reboot`) button to CLI page.

ProDrone, 2015-12-16:
@ledvinap We always reboot after save for consistency. Disconnecting is not the best option IMO. Normally i just click on another tab to see when the FC is back online. Maybe select the first tab when rebooting?

ledvinap, 2015-12-16:
`save` button can easily send "save" command and disconnect board .. much simpler than parsing cli output ...

hydra, 2015-12-16:
i have a solution, committing now.  I think it handles the various scenarious OK for VCP and non VCP boards.

I think a bit of VCP/Non-VCP cleanup is required as in the commit I duplicated some code from the `configuration.js` file.  Likely other tabs that save and reboot all need similar behavior.

jtp755, 2015-12-16:
Thanks @hydra!

hydra, 2015-12-16:
Create new issues and reference this if there are issues with the new code.  I tested it ok a couple of F3 boards I have on my desk and it seems to work fine.

closed 2015-12-16 by hydra

skaman82, 2015-12-16:
Tested with naze rev 5 & 6. Works fine.

commit 76183634 (2015-12-16): Improve CLI mode entry/exit detection. Fixes #285.

Disconnects and reconnects for VCP boards.  After manually saving or
exiting on non VCP boards the setup tab will be shown.

See
https://github.com/cleanflight/cleanflight-configurator/issues/285#issuecomment-165096715