cleanflight/cleanflight-configurator #297: Problem with motors test mode switch and dynamically enabling/disabling switchery sliders

hydra, 2015-12-16:
It doesn't work.

Tried instructions as per the switchery website and issue but to no avail, please investigate.

Basically, if you use an old version of cleanflight that does not support MSP_3D and enable 3D feature the checkbox should never become enabled.

See #233.
See http://abpetkov.github.io/switchery/
See https://github.com/abpetkov/switchery/issues/30
See https://github.com/abpetkov/switchery/releases/tag/0.8.0

See 18fd5d01ec171effb259a4b4689292ad0a0d677c

hydra, 2015-12-16:
@skaman82 Please address this issue when you can.

skaman82, 2015-12-16:
> Assigned #297 <https://github.com/cleanflight/cleanflight-configurator/issues/297> to @skaman82 <https://github.com/skaman82>.

I will have a look in the next view days.

skaman82, 2016-03-15:
Done in #317. Closing

closed 2016-03-15 by skaman82