cleanflight/cleanflight-configurator #305: suggestion: better layout for small screens

huibuuh, 2015-12-21:
I noticed today after my cleanflight updated itself, that it is now almost unusable on my small screen. In the pre 1.x versions I could easily use cleanflight on my small screen in the field. Now I can hardly see things anymore. The thing is that I never need to use clmeanflight configurator at home on my big screen because I cannot fly at home. I only use the program on my tiny windows 10 tablet when I'm out in the field flying.

please change the layout so that it is possible to use the program on small devices outside in the field again!

Robomort, 2015-12-26:
Yes I agree. The pages are too large. The config panel in particular could be spread over a couple of tabs. 
The new colour scheme and little icons are OK on a desktop but all but unviewable on a screen in daylight. Can we please have the option to use the old look and feel.

skaman82, 2015-12-28:
The new configurator allows even little more screen space than the old one an the color scheme is still practically the same in the content areas - it is only dark in the header and menu areas. So here I can't follow you feedback completely regarding the screen space and the colors. The main issue I see in the screenshot is the overlayed menu (scroll bar). This could be done better by making the tabs a bit longer to still able to see the icons clearly.

Robomort, 2015-12-28:
An app from the 90's perhaps but it was clearer and high contrast which makes it usable on mobile devices even in daylight conditions. At the risk of sounding ungrateful but IMHO there are too many pointless changes and BS bells and whistles being added to Cleanflight and it's configurator. Take the new colour scheme for example. It's not even original, just a rip off of Mission Planner.
Please, if it ain't broke don't fix it.

skaman82, 2015-12-28:
@Robomort Thats your opinion and thats is ok. It is simply impossible to make everybody happy by making new stuff. The design is not a rip off of anything. I don't know why somebody would even think that...? It is like saying Ferrari Design is a rip off of Coca Cola. Yes, the brand color of Cleanflight is green (not even the same green) but this is all that the two interfaces have in common. 

A tremendous group of users gave a stellar feedback to the new redesign of the app. I also think that this was a great idea. Yes, it may be not 100% perfect but it is a huge improvement. I don't know what you mean by "BS bells and whistles" but I find that recently added features to configurator are actually very useful and I am hoping that many new useful things will be added in the future.

huibuuh, 2015-12-28:
I think it would make sense to have one variant of the program that is specifically designed to be usable on small devices in the field. Tiny China tablets running full versions of Windows become cheaper and more popular. For drone users they are very useful devices to fine tune parameters in the field while flying. 

That's why I think it would be appreciated by lots of users to have a version of the great Cleanflight that is optimized for tiny screens. Thanks for your consideration!

skaman82, 2016-01-02:
@huibuuh Did some optimization in 
https://github.com/skaman82/cleanflight-configurator/commit/bd7a2fc768f1778523cc1520772ee422fb3768ba (readability: darker fonts)
and https://github.com/skaman82/cleanflight-configurator/commit/b4dc62cfa2ad2b6b326f670c13960536917dc84d (menu: more space for the scrollbar)

skaman82, 2016-01-02:
Looks like this now: 
[image]

Robomort, 2016-01-02:
Definitely an improvement, Thank you.
Making the icons white to improve the contrast would be nice or perhaps highlight the option under the cursor.
I would also like to suggest some rearranging of the page contents to minimise page swapping and logical consolidation of the controls.
1] Reduce the minimum window width  from 1040 to 1024 to better fit small screens.

2] consolidate the Ports and RX pages and incorporate receiver related content from the Config page   this will provide a single page for configuring and monitoring communications to and from the board. The live graph of RX inputs is redundant IMHO and could be removed to provide space for the relocated items.

3] Move the GPS section of the config page to the dedicated GPS page. This is a less used option and occupies valuable screen space on the overcrowded config page.

4] Consolidate the Logging and Flash memory pages into a single page. A page each seems extravagant.

5] remove or reduce the large Cleanfight header from all but the Setup and Home pages to regain usable screen space.

Thankyou for your time efforts and patience.

skaman82, 2016-01-02:
@Robomort 

Thank you for your feedback. Just added a mouse hover effect to the menu icons in https://github.com/skaman82/cleanflight-configurator/commit/bd2f99b4ddeee3ec2b7e258efa3dbbd8dce75f95

Rearranging the contents is not so easy to do.
1. The minimum windows size is 1024 already.
2. Mixing Ports and RX contents is not a good idea in my opinion.
3. The GPS Page will have much more content soon. There is a map already in place in the DEV version and some new config options will follow shortly. So space will be tight here too.
4. This is a good Idea but needs some work and time.
5. The header can not be reduced only for certain tabs at this stage. There are also more elements coming that will be placed in the header. Already placed here a profile switch and dataflash status in the DEV version.

skaman82, 2016-03-15:
No feedback since Jan 2. So I'm closing that one.

closed 2016-03-15 by skaman82