cleanflight/cleanflight-configurator #310: UI Improvement: Duplicate Setup page Battery Voltage in Battery Voltage box of Configuration Page

pfriedel, 2015-12-29:
What the title sez - while tuning out any voltage scale errors it seems odd to have to jump back to the setup page to see what the current scale is reading as instead of tweaking the setting, hitting "Save" and having it update on the Configuration page in the vbat configuration section.

Low-rent mockup below:
[image]

qfather, 2015-12-30:
I think it's a neat idea.  I added the battery voltage to the configuration tab and created a PR, #311 

One small caveat, the voltage does not dynamically update like it does on the Setup tab.  If you're sitting on the Config tab for a while and the voltage changes it won't show this until you cycle tabs or reboot the FC.

[image]

tracernz, 2015-12-30:
You could setup a timeout to update it periodically instead. ;-)

qfather, 2015-12-31:
Not sure if _I_ could. ;)

hydra, 2016-02-05:
Done

closed 2016-02-05 by hydra