cleanflight/cleanflight-configurator #314: UI Improvement: Display 0.01 steps for P instead of 0.1

SoulPictures, 2016-01-02:
Hey there,

within betaflight is seems now possible to increase and decrease P in steps of 0.01 instead of the usual 0.1 steps within cleanflight. Would it be possible to change the display "step" from 0.1 to 0.01. 

It seems as if it would be easily possible within the "pid_tuning.html" file. I yet have no clue how to do it myself and commit it to the project, i am still to new to github for this sorry.

[image]

skaman82, 2016-01-02:
@SoulPictures This is not quite so easy. In the js. file is some weird math rounding action going on. This was too tricky for me as I looked into this.

tricopterY, 2016-01-03:
This becomes significant if you are using pidLuxFloat Controller as per betaflight i.e calculations take decimal points into account. Otherwise all the PID terms are calculated/rounded as whole numbers (integers), and fractions/decimal points have no meaning. 
Summation of PID terms is always in whole numbers.
This is all about Resolution of PID terms and their real/over-all effect.
Happy flying.

NightHawk32, 2016-01-12:
Greater resolution also requires firmware changes, there is a discussion ongoing: https://github.com/cleanflight/cleanflight/issues/1675

martinbudden, 2016-05-08:
In betaflight`pidLuxFloat` has been converted to use PID integers values on the same scale as `pidMultiWiiRewrite`.
The same change is happening in Cleanflight v1.13.0.

So all of these steps should be set to 1 and max ranges set to 255.

See PR https://github.com/cleanflight/cleanflight/pull/2016

closed 2017-02-25 by hydra