cleanflight/cleanflight-configurator #329: Manifest Missing or Unreadable

JoeMB84, 2016-01-29:
While trying to update Cleanflight to its latest version (1.11.0) via the extensions in Google Chrome, I receive this message. Am I missing something?
[image]

tricopterY, 2016-01-30:
You can only update the Flight Control (FC) firmware via the Configurator. You already have the Configurator installed as an extension. So, run that and choose Firmware Flasher and let it load all the available FC board options, and then select your appropriate board, before clicking on Load Firmware (online).

JoeMB84, 2016-01-30:
Oh, I see. So, this is for an update of the firmware for the FC, not the configurator? Sorry for such a noob question. LOL!

closed 2016-02-05 by hydra