cleanflight/cleanflight-configurator #340: Add Acro Plus to the inflight adjustments (acro_plus_factor), plus other improvements.

Ozzy33, 2016-02-12:
Add Acro Plus to the inflight adjustments (acro_plus_factor).

GUI allows 3 profiles, labeled 1/2/3 (in top right corner of the gui), but in CLI it only allows 2 profiles and they are called 0/1, kinda confusing imo.

Add all possible modes to modes page, greyed out when not able to select and popup text explaining what to set to enable that mode.

CLI search box.

Sensor tab has new debug graphs, add text to expalin what these are:
Debug0=Looptime,
Debug1=Jitter?
Debug2=?
Debug3=?

When might we be able to backup CF/BF with just a cli dump or better a single button save/restore?

The new CFC v1.2 is great, alot of new items, excellent work, thanks!

skaman82, 2016-03-15:
@Ozzy33

• AcroPlus is still experimental - if it sticks around, it maybe makes sense
• The are two different profile groups (profile 0-2/rateprofile 0-2) since the beginning - yes it is confusing and worth of sorting out 
• Modes: that is probably a nice idea, it would add a lot of scrolling space, tho. Thoughts?
• CLI: Cli will be removed hopefully soon
• Debug Labels = good (I still don't know what they are for)
• Backup: This can be done only after the CLI is gone and a new parameter handling interface is in place. So we have to wait a while for this.

fftunes, 2016-03-19:
betaflight has 2 profiles, configurator shows 3. As the cleanflight number of profiles is just a number and could be set to 4 or 5 or whatever when compiling, would the configurator support 'n' profiles? I fear not.

Modes greyed out is good idea. Popup aid too.

CLI can kind of search already. For example type "set yaw" and it will show any variable having "yaw" in its name.

Why should the CLI go away? How would it help not having it? It is so useful in development for trying new stuff, how could you possibly make the GUI make up for that?

closed 2017-02-25 by hydra