cleanflight/cleanflight-configurator #342: Port setting "LTM" is not saved in Cleanflight-Configurator 1.2.0

HenningFlyer, 2016-02-13:
I flashed "iNav 1.0 release" to my FLIP32 10DOF board today, then tried to set up with CF Configurator 1.2.0, just as I did before with older FW versions on this FC before:

CONFIGURATION:
 - Softserial ON
 - Telemetry ON
 - GPS ON

PORTS:
 UART1: MSP
 UART2: GPS - 57600Bd  (using a UBLOX neo m8n)
 SOFTSERIAL1: unused
 SOFTSERIAL2: **LTM **- 9600Bd (using a DIY converter to Jeti EX telemetry)

The "LTM" setting can be selected, but after saving and rebooting the FLIP32, the dropdown field jumps back to "disabled" again. To find out if its a íNAV FW issue or a CF configurator issue I then tried to do the same with EZ-GUI on my Android Phone...  ==> With EZGUI, I suceeded to save the LTM setting! (tested OK in flight already)...

It seems to me like the latest Cleanflight GUI does not handle the "LTM" function of the ports correctly...  Or is this some incompatibilty between CF GUI and iNAV FW?

Did anybody else already suceed to set "LTM" with current Cleanflight GUI v1.2.0?

Kind regards,
   Henning

DzikuVx, 2016-03-14:
I have exactly the same problem here. Configurator 1.2, CF 1.12. Unable to set LTM

BaronZamboni, 2016-04-09:
Same problem using Configurator 1.2.1 to CC3D running firmware 1.12.0
LTM telemetry option is not saved on reboot.

BaronZamboni, 2016-05-05:
I was doing some more reading on this and it appears that on March 24th issue #355 was closed with a fix for this same problem. 

I've looked at the code change added to js/msp.js and see the fix that was put in for change a924e06 reflected in the current source.

The problem is still affecting some boards.  Hydra would you please re-test this using a CC3D board?  I'm using firmware 1.12.0 and Configurator 1.2.1 and I cannot save the LTM telemetry option on the port setup.  I have the telemetry option enabled and that setting does save successfully.

https://github.com/cleanflight/cleanflight-configurator/issues/355

https://github.com/cleanflight/cleanflight-configurator/commit/a924e06c1bd8a28479b4d0ab33086920690ba943

closed 2017-02-25 by hydra