cleanflight/cleanflight-configurator #349: Need setup wizard for motors/receivers settings

proggod, 2016-02-28:
Is there any issue with me creating some sort of a wizard, so it automatically sets up mmix settings, rxrange settings, midpoints, receiver types, etc?   Seems it would be a lot easier for someone to use cleanflight, if there was a simple wizard to do all these settings.  It took me a few days to master all these settings, when it could easily be done automatically?

Is the best place to talk to someone about getting up on the code base, irc?   If so who should I talk to?

alanzanotto, 2016-03-01:
this would be awesome

proggod, 2016-03-01:
I think I will start working on it tomorrow, if no one else has any objections, or any ideas how they think it should be done..

skaman82, 2016-06-16:
duplicate of: #98

closed 2016-06-19 by proggod