cleanflight/cleanflight-configurator #350: Spektrum TAER Config

kewlkeith, 2016-03-04:
First off, thanks for all the awesome work that has done! 

Is it possible to configure CF to automatically select TAER when choosing a spektrum2048 or spektrum1024 serial rx option? 

Thanks,
Keith

fftunes, 2016-04-01:
2 cents:
- it would probably make TAER the new default, since spektrum1024 is set default wether you use serial or not
- guessing it would be easily possible, but IMHO autoconfigurations across the tabs are risky if they go unnoticed

zombiehugs, 2016-08-31:
Newer Cleanflight user here, from a UX standpoint I did find it a bit odd that Configurator did not automagically select the proper profile based on my previously selected receiver profile. Based on @fftunes comment I still have some discovery to do in the source as to how auto configurations can or don't work in configurator.

closed 2017-02-25 by hydra