cleanflight/cleanflight-configurator #351: cli beeper dump output is not pasteable

Hydroculture, 2016-03-04:
Hi!

Not a big problem.

When doing a dump, the part saying
# beeper
beeper  GYRO_CALIBRATED
beeper  RX_LOST
beeper  RX_LOST_LANDING
beeper  DISARMING
beeper  ARMING
beeper  -ARMING_GPS_FIX
beeper  BAT_CRIT_LOW
beeper  BAT_LOW
beeper  -GPS_STATUS
beeper  RX_SET
beeper  ACC_CALIBRATION
beeper  ACC_CALIBRATION_FAIL
beeper  READY_BEEP
beeper  MULTI_BEEPS
beeper  DISARM_REPEAT
beeper  ARMED
beeper  SYSTEM_INIT
beeper  -ON_USB

can not be pasted. There is one space too much after "beeper"

May be you can fix it?

Thank you 

best
Holger

Pierre-A, 2016-03-18:
**@Hydroculture** ,  ooops my bad, not related to Configurator.
Solved [here](https://github.com/cleanflight/cleanflight/pull/1894#issuecomment-198333995)

Pierre-A, 2016-03-18:
So you can close this issue..

closed 2016-03-24 by hydra