cleanflight/cleanflight-configurator #357: Configurator should check board type before flashing

MasterZap, 2016-03-17:
When you flash, the configurator sends boot command to the board and proceeds to flash it.

What if it first sent a "version" command to check what board it was, and proceed to issue a warning if you were trying to - for example - by mistake flash the RMDO firmware to a Naze32 mini. :)

That would be awesome, wouldn't it?

(And yes, I confess, I did exactly that mistake last night :) )

/Z

proggod, 2016-03-17:
Would this even work, since when your flashing your in DFU mode since your flashing the boot loader.

MasterZap, 2016-03-19:
But aren't you sneding the command to reboot to DFU mode? The idea is to check BEFORE doing that?

proggod, 2016-03-19:
i am putting it in DFU mode before I even plug it into the machine.

MasterZap, 2016-03-19:
Right but then you need to turn on "no reboot sequence", right? This would be part of said "reboot sequence"

proggod, 2016-03-19:
Honestly it works either way, for sparkys or CC3Ds i short the bl pins, then connect to computer, it recognizes as DFU mode, and flashes.   I don't think there would be any software to know what type of board it is, or what is already on the board at this point, sinces its doing a raw write to the boot loader.   I could be wrong though..

MasterZap, 2016-03-20:
But who actually shorts booting pins to flash? I, and everyone I know, connect it "normally" and just goes to the flashing page and hits flash?? That reboots the board into DFU mode and flashes it. But *before* it does that, it could just query the board for the version string, and do some sanity check... so if the firmware on the board says "NAZE" and the firmware you are trying to flash says "RMDO" maybe it should, like, cough gently? :)

MasterZap, 2016-03-20:
Specifically: When flashing with the "no reboot sequence" switch OFF, *before* sending MSP_REBOOT command, the configurator should send MSP_BOARD_INFO and get the board name, and check that that string exists in the filename being flashed, and if not.... pop a warning.

hydra, 2016-03-24:
@MasterZap yes, this is EXACTLY why I had conversations with Alex and Bart (from EZI-GUI) when i met up with them in Paris over a year ago.

Changes were made to the API to support this functionality.  Wanna have a go at it?

MasterZap, 2016-03-28:
I've gotten as far as downloading the configurator code but Life has intervened so far (illness, stuff, and too nice flying weather) :)

But it's on my infinitely long list to try to do.... :)

MasterZap, 2016-03-28:
(But I've done nothing so far... so if someone else wants to have a go, feel free). 

My idea was to add a new "Board Detection" switch (that would be grayed out and off if "no reboot sequence" was on). 

This switch would suck in the board 4 letter code, and have a mapping from that to the target (file) name. 

It would then:
- Filter out anything not containing that name in the drop down of firmwares to pick
- Complain with an "are you sure what you are doing" dialog if you pick a downloaded local file that doesn't contain the string.

MasterZap, 2016-04-08:
So I've dicked around a little with this but can't get it to work :(

How can I send the MSP command to detect board type? I tried this, in stm32.js, just before it sends the reboot message. This is test code, so it returns at the end and doesn't actually flash anything... lines mareked with ">" are existing lines, the unmarked are my added lines....

>            if (openInfo) {
>                // we are connected, disabling connect button in the UI
>                GUI.connect_lock = true;

                if (self.options.detect_board_type) {
                    // Send board type identification message

                    serial.onReceive.addListener(MSP.read);
                    MSP.send_message(MSP_codes.MSP_BOARD_INFO, false, false, get_board_handler);

                    serial.read();
       
                    function get_board_handler() {
                        board_definition = BOARD.find_board_definition(CONFIG.boardIdentifier);
                        console.log('Using board definition', board_definition);
                    }
                }

                serial.disconnect();

                GUI.connect_lock = true;

                return;

>                console.log('Sending ascii "R" to reboot');
>                var bufferOut = new ArrayBuffer(1);
>                var bufferView = new Uint8Array(bufferOut);

Help? :)

closed 2017-02-25 by hydra