cleanflight/cleanflight-configurator #361: Add min_check and max_check GUI to Receiver tab

adamweld, 2016-04-07:
Hey fellas, first off THANK YOU for the great work you've been doing. The cleanflight configurator is a great piece of software.

This idea came to my head last night - it's a fairly common problem for new users to complain that their quad doesn't arm, even though everything else seems to be working perfectly. The most common cause of this is that their radio isn't set up to send values with high enough magnitude to pass beyond min_check/max_check values. Sometimes this can be fixed by changing radio settings, but cheaper radios don't have this capability. Thus the aforementioned variables must be edited in the cli.

My idea is to add to the receiver tab an intuitive visualization of what min_check and max_check do, and a way to change their values. Here's a quick mock up in paint.net:
[image]

This way, when you're moving your sticks around in the tab it's easy to see that they don't cross into the "check zone".

I don't have any real experience making chrome apps, but if someone is willing to give me some pointers I'd love to attempt to add this feature myself. Keep up the good work!

skaman82, 2016-05-06:
This requires just some HTML/CSS/JS work - so if you're up for it just give it a try.

gaelj, 2016-05-08:
You just need to use the "load non-packed extension" button in Chrome's extensions page to setup your local dev files as a chrome app. Click on launch in chrome the first time, then pin the app to the taskbar and you can open it directly.
You might have to do right-click / update application instead of normal kill / restart to apply some code updates (mainly for changes in the language files).
You can use the Chrome developer features to debug, for example step-by-step execution or live modification of css parameters.
I personally use github desktop to sync, and eclipse IDE for dev.
Your idea looks fairly easy to implement, you should manage to something up just with existing code study and some googling your way around. You should start by creating your own fork of CF Configurator, commiting your code changes to it. Then when you are happy with it you issue a pull request.

closed 2017-02-25 by hydra