cleanflight/cleanflight-configurator #375: Configure ESC parameters directly

ledvinap, 2016-04-28:
https://github.com/cleanflight/cleanflight/pull/2075 and https://github.com/cleanflight/cleanflight/pull/2090, it is easy to implement 4way interface over MSP.

With it, it would be possible  to configure ESC parameters directly from configurator.

Is there someone who can work on configurator part? I can implement firmware support.

4712, 2016-05-01:
I have thought about, but I'm afraid, my coding style is not cleanflight compatible...

ledvinap, 2016-05-02:
@4712 : Sorry, I didn't mean to offend you. I wanted to redesign the entry from MSP only. But it got out of my hand a bit..

skaman82, 2016-05-06:
@ledvinap I can help you out with the GUI layout and html/css - with the JS someone else has to take over

DieHertz, 2016-07-27:
Do we really need to redesign anything? My idea is to do all the work in CleanFlight Configurator, say add  a built-in "BLHeli Configurator" limited in terms of supported interfaces.
We basically have to implement a part of BLHeliSuite functionality, allowing to at least read/write EEPROM and map all the settings.
I can do the JS and C++ (if required) parts, you can help me with the presentation layer. I'm currently figuring things out in my local repo, will check back in later.

DieHertz, 2016-07-29:
I'm currently working on it in my own repo. Suggestions about GUI are highly desired and appreciated.
Are we going to mimic BLHeliSuite by displaying all available settings, or some restricted subset of them? What about different modes? I suggest initially supporting only MULTI.

closed 2017-02-25 by hydra