cleanflight/cleanflight-configurator #382: Prepare for configurator 1.13.0 release

martinbudden, 2016-05-13:
### TODO

1. Create release notes
2. Wait for bug reports and make a list of them here.

### Release Candidate bug reports

### DONE
1. None yet

hydra, 2016-05-13:
Need to investigate why current sensor reads 65536 in latest configurator.  i noticed this when testing today.  not sure if it's MSP or firmware related yet.  Have someone else confirm this would be good.

reedbn, 2016-05-15:
Issue #384 needs to be addressed

skaman82, 2016-06-17:
@martinbudden here a list with things to do:
https://github.com/cleanflight/cleanflight-configurator/pull/362
https://github.com/cleanflight/cleanflight-configurator/pull/397
https://github.com/cleanflight/cleanflight-configurator/pull/399
https://github.com/cleanflight/cleanflight-configurator/pull/373
https://github.com/cleanflight/cleanflight-configurator/issues/381

NEW OSD Tab (because of that I would say we jump right to 1.3 instead 1.2.3.

Or maybe it is worth it to release 1.2.3. in the current state (that way users can benefit from advanced LED tab before the OSD stuff is ready (this can take some more weeks I think). @hydra What do you think?

martinbudden, 2016-06-17:
Personally, if you are confident in the code, and you think you haven't introduced any bugs, I'd release 1.2.3 now.

skaman82, 2016-06-17:
Yes, I am pretty confident that i didn't break anything :)

skaman82, 2016-06-17:
@hydra I am also for releasing 1.2.3 now and then start to working on 1.3.0

closed 2017-02-25 by hydra