cleanflight/cleanflight-configurator #383: PID values should be scaled 0 to 255

martinbudden, 2016-05-13:
Currently the PID values are inconsistenly scaled, currently we have (taken from `tabs/pid_tuning.html`)

For roll, pitch and yaw we have:

PID | min | max | step
---|---|---|---
P | 0 | 25.5 | 0.1
I | 0 | 0.255 | 0.001
D | 0 | 255 | 1

Now that `PID_LUXFLOAT` has been converted to use the same range as `PID_MWREWRITE` these should all be scaled to `min:0, max:255, step:1`. Corresponding changes need to be made in `tabs/pid_tuning.js`.

skaman82, 2016-06-16:
@gaelj i think this is a relative quick fix. Could you have a loot at it if you have some time? It also needs a fallback for older API versions.

gaelj, 2016-06-16:
As I understand it this is not due to a change in the MSP protocal.
In this case msp.js seemed a more appropriate place to do the change, as that is where the original scaling is applied.
Backwards compatiblity is implicit - I did not change the MSP content, just the way it passed back and forth from MSP to configurator.

martinbudden, 2016-06-17:
@gaelj , that is correct, this is not due to a change in the protocol. It's due to a change in PID_LUXFLOAT which now uses the same integer values as PID_MWREWRITE. As such it makes no sense to use fixed point values. This is really just a question of how the data is displayed in the configurator - the data passed back and forth is still the same.

skaman82, 2016-06-17:
@martinbudden this looks good to me – but there is no fallback for oder CF versions

[image]

martinbudden, 2016-06-17:
Looks good to me. I'm not sure that fallback is necessary, after all it's only a move of decimal point, which shouldn't really confuse anyone.

skaman82, 2016-06-17:
done in https://github.com/cleanflight/cleanflight-configurator/commit/8c37bdfa5cd7dc116a66649a9b569720ea472c31

closed 2016-06-17 by skaman82