cleanflight/cleanflight-configurator #384: Potential VCP issue in 1.2.2?

Arakon, 2016-05-15:
I noticed that while I have zero issues with several flight controllers running a CP2102 bridge to UART, my VCP controllers (CC3DF3 and SPRacingF3 Evo) cause severe lag, even the green link icon occassionaly turns off briefly, sensor readings especially on the motor page skip and sometimes freeze for up to two seconds. This is independant of the looptime set on the controller (CPU load of 3%).
Originally thought it was just me, but someone else reported something similar on a forum, so I figured I'd open an issue in case there is a problem with the configurator.

reedbn, 2016-05-15:
I bumped to 1.12.0 firmware on CC3D, and 1.2.2 configurator and had this exact issue - super laggy, even on the landing page showing the orientation of the quad. Dropping back to 1.11.0 and appropriate configurator resumed normal, non-laggy behavior

Arakon, 2016-05-16:
I'm on betaflight 2.7.0, so I'm fairly sure it's not directly related to the controller software, but the configurator itself.

gaelj, 2016-05-16:
I don't have a VCP board (yet). Could someone please try out this fix ?
https://github.com/gaelj/cleanflight-configurator/commits/VCP-lag

Arakon, 2016-05-16:
Didn't see any difference really. On motor tab it may be a little smoother, but on sensor tab, looptime still jumps bigtime and it lags.

hydra, 2016-05-16:
@reedbn please try 1.13.0-RC1 and report back when you have tried it please.

gaelj, 2016-05-16:
I've made a new commit of the configurator here https://github.com/gaelj/cleanflight-configurator/commits/VCP-lag where live status update is disabled (the new box with battery, armed status etc won't show).
I want to know for sure if the issue comes from this change or not.
Could you please try this version both with Betaflight and CF on your VCP board ?

reedbn, 2016-05-17:
Just tried v1.13.0-RC1 and it seemed to all behave nicely with configurator v1.2.2.

hydra, 2016-05-17:
@reedbn ok thanks for the feedback. good to know.

I think we should just mark this as fixed and move on.  Thoughts @gaelj ?

gaelj, 2016-05-17:
@hydra I've added a last commit to the branch here https://github.com/gaelj/cleanflight-configurator/commits/VCP-lag
Could you take a quick look into it ? It's not much but I do think it is an overall improvement, albeit small.
If you agree then I'll create a PR based on it, or I can improve it more if you see anything that needs to, or just let it go - up to you.

reedbn, 2016-05-17:
I'd like to hear @Arakon 's feedback first. If 1.13.0-RC1 works for @Arakon too, close this and move on. If not, more investigation is in order.

gaelj, 2016-05-17:
True @reedbn although he seems to be mainly having problems with Betaflight. As using the latest CF fixes the issue, I would tend to think it's up to Betaflight to adapt.
So @Arakon please check how BF behaves with CFC with this commit https://github.com/gaelj/cleanflight-configurator/commit/cb4c99bf92b5e9d27304d406a383fb67d7a5f245 (which disables live update) and then with the latest commit of the fork (which re-enables it with lower frequency and small improvements)

hydra, 2016-05-17:
@gaelj perhaps attach 2 zip files one for each version you want him to test, in case @Arakon does not know how to use git/github.  btw, you can download a zip from github of a specific version via the github UI.

gaelj, 2016-05-17:
Absolutely.
- [cleanflight-configurator-disabled_live_update.zip](https://github.com/cleanflight/cleanflight-configurator/files/268965/cleanflight-configurator-disabled_live_update.zip)
- [cleanflight-configurator-VCP-lag-latest.zip](https://github.com/cleanflight/cleanflight-configurator/files/268966/cleanflight-configurator-VCP-lag-latest.zip)

Additionnally, I received my HK SP Racing F3 Clone today (don't hit me too hard, a genuine EVO is on it's way and this one will be a nice development platform. Quality is average, I broke off one of the connectors after 5mn. Managed to resolder it though).
Anyway, so I flashed it with BetaFlight 2.7.1 (2.7.0 is not easily available as a hex AFAIK). It works without any lags at all, whether on the regular dev branch of CFC or on the latest commit of my VCP branch.

Arakon, 2016-05-18:
Will test later tonight.

Arakon, 2016-05-18:
Hrm. I can't reproduce it currently at all, with BF or CF. I'll keep an eye on it to see if it occurs again. It may have been a memory leak or something, as I rebooted since and it seems to be gone since then.

Arakon, 2016-05-23:
It appears to be related to the extended blackbox log viewer. When that has been running for a while, the CF configurator starts lagging. So not directly related.

closed 2016-05-23 by Arakon

hydra, 2016-05-27:
@Arakon great.  did you raise an issue in the blackbox log viewer issue tracker or otherwise report that?  If not please do.