cleanflight/cleanflight-configurator #385: First flash attempt always fails on naze32

gaelj, 2016-05-16:
I always need to press the flash button twice to flash both my naze32 full rev6, and naze32 acro rev 5 boards.
The first time I press flash after a boot-up, I always end up with the error "Communication with bootloader failed" (on the rev 6 board) or "STM32 timed out" (on the rev 5 board). Then if I press the button a second time, flashing is fine. I get this every time I want to flash.
Does anyone else have this behaviour ?
Any ideas as to what could cause this ?

Arakon, 2016-05-16:
Same happens on SPRacing F3 here.

skaman82, 2016-06-16:
For SPracing F3 it is important to select manual baud rate 256000. Had this problem too occasionally. Reconnecting the usb cable solved this for me every time.

gaelj, 2016-06-16:
Manual baud rate does work indeed around the problem. Then we should maybe look into changing default baud rates for certain boards. I have Naze32 r5 & r6, SPR F3 Deluxe and SPR F3 Evo. It would be nice if people with the same issue on other boards could post their working baudrates in this thread.

Arakon, 2016-06-17:
Not here. I have the default baud rate set to 256000 and it still happens about 1 out of 3 flashes. It works fine nontheless.

closed 2017-02-25 by hydra

ehlsm, 2019-06-20:
the lowest baud rate works for me

kushalarya, 2020-05-10:
I was able to resolve, follow these steps - 
1. Install [STM32 driver](https://www.st.com/en/development-tools/stsw-stm32102.html)
2. Download [.hex file](https://github.com/cleanflight/cleanflight/releases) for your board, in my case NAZE
3. Jump Boot pins on your board. 
4. Connect it to computer. Use STM32 driver to load hex file into board. 
5. When done flashing, disconnect and connect usb ( not need to jump boot pins )
6. Open Cleanflight, check comm port, Click connect 

If it doesn't connect on cleanflight, try flashing older version of hex file.