cleanflight/cleanflight-configurator #392: GUI Backup does not restore modes

Eloo, 2016-06-09:
Hey folks,

I just upgraded my Cleanflight to version 1.13.0 and saved/restored my configuration using the GUI Backup feature.
After the restore I noticed, that my configured modes were not restored for the Profile 2 and 3.
The modes for profile 1 were restored successfully but profile 2 and 3 is empty.
So I checked the backup file and found all my modes for all three profiles, so I assume the backup works properly and the issue is related to the restore process.

To reproduce the problem backup your config(with configured modes in profile 2 or 3) via GUI and reset your configuration in the setup tab. 
Restore your config using the GUI restore function and check the modes tab for profile 2 or 3.

I hope somebody can confirm the problem or give a workaround.

Best regards
Eloo

hydra, 2016-06-11:
the old backup system needs to be replaced with something that backs up all paramater groups (see firmware v1.13.0)  this will fix the problem.

hydra, 2016-10-29:
closing, backup / restore functionality removing pending new implementation.

closed 2017-02-25 by hydra