cleanflight/cleanflight-configurator #403: Roll, Pitch and Yaw rates not correctly set when changing Profiles 1 or 3

xoga74, 2016-06-24:
When selecting profile 3 (=rateprofile 2)  or profile 1 (=rateprofile 0) in configurator, setting the rates in the GUI changes rates in both profile 3 and profile 1.

Example.  : switch to profile 1, set rollrate to 0.5, save,
 then switch to profile 3, the reading gives allready 0.5, set rollrate to 0.9, save
 then switch back to profile 1, reading gives 0.9

When setting the same rates in cli, everything is fine and rates can be setup individually for all the profiles...

skaman82, 2016-06-24:
Just tested this with CF 1.13 and newest configurator. Works for me with no problems

xoga74, 2016-06-24:
I tested it again, same problem, so uninstalled configurater from chrome and reinstalled.  Now working fine..  Thx Closing the issue..

xoga74, 2016-06-24:
Hello, i was able to recreate the problem.  The problem occurs when editing is done with battery power on.

Tried it on different quads, same problem.  Turned on external power and only power on USB, no problem changing.

Can you test if you get the same ?

hydra, 2016-08-24:
rate profiles were never properly handled in the GUI, the real issue here is to provide proper support. There was one bug relating to this in the current version which has been fixed in the latest development version so please try that.

closed 2016-08-24 by hydra