cleanflight/cleanflight-configurator #416: Request for target : Pixracer F427VIT6

Maxzor, 2016-07-28:
Hello, I do not know where to start.

Are HAL and F4 support in 3.0 release?

What is needed to run betaflight on this board? I push it cuz i have it obv, but more important because it could serve as a great hybrid platform between APM/PX4 stacks and true racing stacks among which BetaFlight.

Can provide specs, don't know which are needed tho.
http://www.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32f4-series/stm32f427-437/stm32f427vi.html

BR Max

skaman82, 2016-07-28:
Hi Max,

You posted this in the wrong branch. You should post your questions regarding betaflight in the  betaflight repo and close this issue here. Or better use rcgroups for questions.

gaelj, 2016-07-28:
@Maxzor can you close this please.

Maxzor, 2016-07-28:
Sorry and thumbs up for the Cleanflight project even if im willing to help BF atm :)

closed 2016-07-28 by Maxzor