cleanflight/cleanflight-configurator #419: Upgrade warning

richard-scott, 2016-08-19:
Why was this added: 

https://github.com/cleanflight/cleanflight-configurator/commit/aba4569e506223d6822ff083c43013b981a6cd95

It's really confusing people to be asked to upgrade when there is nothing to update.

JohnOCFII, 2016-08-19:
At the time that code was added, there were new releases every couple of weeks, and there were a lot of support questions by people who were running old code.  This was put in place to help ensure fewer people would run into issues based on using an out-of-date configurator.  You'll note that that particular change took the upgrade message from 2 months to 3 months.  So the pass of change is certainly less, and it is confusing.  It would be best if there were some way to actually check for an update, and only alert in that case, but I don't know if this is made more difficult for Apps in the Chrome store ecosystem.  Ideally, one of the current application developers will step in with a comment too.

richard-gtv, 2016-08-22:
Thanks for the info @JohnOCFII, I agree, it would be nice if it only alerted when there actually was an update :-)

hydra, 2016-08-24:
i think we should just delete that tbh.  I'm going to do it now.

hydra, 2016-08-24:
done, see 530bdad1df61a18530169e520f060ca3e104d141

closed 2016-08-24 by hydra