cleanflight/cleanflight-configurator #420: Google Chrome apps are no longer being supported

Austinthemighty, 2016-08-21:
here is a link to the article that state that google is dropping support for google chrome apps: http://wccftech.com/google-planned-support-chrome-apps-mac-windows-linux/

In summary;
Google to phase out Chrome apps on desktops means that users will no longer be able to run Chrome apps on any platform except for Chrome OS. Google’s plans start later this year as Chrome apps will only be available to Chrome OS users. Current installed apps will not be affected by this and can still be updated. Moving forward at sometime in the year 2017, the Chrome Web Store will no longer display Chrome apps at any of the above mentioned operating systems. Finally, in 2018, you will not be able to load any of them on Macs, Windows and Linux.

one thing for sure is we need to start to move away from the google chrome app store, also maybe a dedicated app that you download, it may take some time to do this, that is why i was putting this here.

bripkens, 2016-08-22:
Not sure whether you already have ideas about this, but [Electron](http://electron.atom.io/) might be an option to reuse almost all of the code, i.e. eliminate the need for a large rewrite. As a bonus, the app would remain cross platform and would even work for folks who don't like Chrome.

hydra, 2016-08-23:
does electron have USB and Serial API's?  I couldn't see them here:

http://electron.atom.io/docs/api/

bripkens, 2016-08-23:
@hydra: You can use the full Node.js API. Meaning: You will certainly be able to use USB and serial APIs. There is probably a Node.js library out there that fits cleanflight's / betaflight's needs. Anything you specifically need (I never build something which talks to stuff via USB).

hydra, 2016-08-23:
Any research you can do regarding this would be most welcomed.  Links to other projects/examples/libraries would really help.

ledvinap, 2016-08-23:
Waiting may be better strategy - other projects will face this problem to, so proven solutions will be around soon

Austinthemighty, 2016-08-23:
why wait, if we wait it may not get started tell mid 2018, i feel like this is a good thing that i came across, also i did talk to @borisbstyle and he said that beta flight is already on the move and has been for some time now!

etropic, 2016-08-23:
"...Electron accomplishes this by combining Chromium and Node.js into a single runtime..."
Is that really the way to go considering Chrome changes?

Also, a quick glance of links from electron to node for usb support ends with requiring generic drivers with http://zadig.akeo.ie/
Probably warrants a deeper dive.

ambrosechua, 2016-08-24:
http://nwjs.io/blog/whats-new-in-0.13/

Austinthemighty, 2016-08-24:
ok, looks good, will give a test run tomorrow and report back with my findings, also i hope there is a way to have like auto updates when  ever there is an update

hydra, 2016-08-24:
yeah, tbh the configurator codebase sucks anyway, i've never been a fan of jQuery since I first used it over 10 years ago.  I'm happy to see the old codebase die and just used for reference of how not to do it.

hydra, 2016-08-24:
In the short term I we can update the configurator to meet current requirements, but I certainly don't see a whole lot of point attempting to rescue the codebase via angular/js or similar testable frameworks any more.

Austinthemighty, 2016-08-24:
ok, what language should we use

Austinthemighty, 2016-08-24:
@hydra also go ahead and create a new repository for the new configuration

bripkens, 2016-08-24:
I'd highly interested in helping build up a new configurator UI as well!

etropic, 2016-08-24:
@Austinthemighty I guess first question is what are the current design goals? Is it just cross platform targeting, or are there other lessons learned that would help define new goals?

Austinthemighty, 2016-08-24:
@etropic my current goals are

1. is to get it to no longer run on google chrome
2. have it be all platform compatible, except for chrome os( not sure what to do for them)
3. also create a ios/android program that is able to communicate over blue tooth
4. make it look and feel like the current configurator.

Austinthemighty, 2016-08-24:
@bripkens i am going to need all the help i can get, i am currently going to school from 7:15am to 6:30 pm

etropic, 2016-08-24:
@Austinthemighty 
    cool, #3 we can add win phone as well (it would be worth it for me at least).
#4 I like the way the sliders and controls work, but always though it could be better regarding the page navigation on the left column, maybe there are some improvements that can be made.

JohnOCFII, 2016-08-24:
Ahh - requirements!  This may take a while.  :) Time spent here will be well worth it down the road.

1) Cross platform -- for sure, but how best to define?  Still browser-based?  App-based?  If this can still work in a browser, I'd have to believe the on-going support would be less effort than using something like QT (https://www.qt.io).  Even within Windows, Mac, Linux, Android, and iOS, it is probably fair to set some limits (no support for something older than 5 years, for example).
2) An eye towards maintainability and documentation.  One of the most difficult things in most open source projects is to keep them alive and documented year after year.
3) An eye towards sharing.  Currently CleanFlight, BetaFlight and iNav all use a configurator based on the one from BaseFlight.  If significant effort goes into creating a new configurator, it would be great if it could support the current (at least at this time) flight controllers as listed above.
4) Easy to use.  Good UI/UX - not just pretty buttons, but logical groupings and flow to configuration settings.
5) Some level of self documentation.  I'm a fan of short help "bubbles" that pop up when hovering over a configuration item.  I'm sure there are other good ways to do this too.

Just some random thoughts from a user who used to do software development 20 years ago.

Good Luck!

Austinthemighty, 2016-08-24:
@bripkens @hydra @ledvinap @etropic  @ambrosechua @JohnOCFII ok this sounds good i am going to start a new repo and add all of you to the members, and also once @hydra is ready i will transfer the ownership to him

Austinthemighty, 2016-08-25:
i have sent all of you invites to the repo and slack, @etropic @ledvinap i was not able to send you invites because i didn't have your email.

etropic, 2016-08-25:
etropic at tropicdesert dot com

bripkens, 2016-08-25:
How would our effort be synchronized with https://github.com/betaflight/betaflight-configurator? Do we need to synchronize it or would a new configurator be based on the betaflight-configurator? (cc: @borisbstyle / @mikeller)

Austinthemighty, 2016-08-25:
i don't think we will sync up with betaflight, but we might, i have already got the ball rolling on the same issue, but @borisbstyle might piggyback of of this, also we might integrate a lot of the betaflight stuff into here, not entirely sure yet

hydra, 2016-08-26:
no more slack channels please!  There's far too many messaging systems for me already.  People contact me via 4 different email addresses, 2 twitter account, 4 slack channels, 3 facebook accounts, 3 facebook pages, 6 instant messanger accounts, 2 phone numbers and sms numbers.

Austinthemighty, 2016-08-26:
so who knows x-code well

zombiehugs, 2016-08-31:
@Austinthemighty, keep me in the loop on this development and add me to whatever communication and repo you guys set up. I'm looking to lend a hand on configurator refactor. Primarily front-end developer here, but am comfortable in all areas of the stack.

Austinthemighty, 2016-08-31:
@zombiehugs do you have an email that i can send you an invite to
also here is the repo: [Cleanflight-Configurator 2.0](https://github.com/Austinthemighty/cleanflight-configurator2.0 )

tcmichals, 2016-09-02:
What will be the basic GUI technology QT, Web or something else? Programming language, C++, Javascript?

Austinthemighty, 2016-09-02:
@tcmichals i was planing on creating the os x and ios in xcode, and the windows in Visual Studio

etropic, 2016-09-02:
We could use Xamarin for front end across all platforms. I think that would leave a very small platform specific piece for the serial data.

tcmichals, 2016-09-02:
@Cleanflight-Configurator 2.0, there are 3 projects, Windows using C#,  QT UI forms (C++), IOS using Swift.  So the goal is to Configurator in 3 different languages?

tcmichals, 2016-09-02:
@etropic  I think that be an issue on native Linux https://forums.xamarin.com/discussion/59305/xamarin-forms-on-linux 
Xamarin only support IOS, Android, and Windows.

etropic, 2016-09-02:
@tcmichals ya, I was hoping that it would mostly be covered in mono, but reading a bit it may be a bigger gap than I admittedly assumed.

Austinthemighty, 2016-09-02:
@tcmichals as of right now yes, but that is still up in the air, also i think i can make the ios and the os x app all at once.

Austinthemighty, 2016-09-02:
@tcmichals do you have an email so i can send you an invite to slack

tcmichals, 2016-09-02:
@Austinthemighty  tcmichals at yahoo dot com

ambrosechua, 2016-09-03:
By the way, to run this in NW.js, I only had to write [this](https://gist.github.com/ambrosechua/5c689c2311ab7f22dbd1b5e67363cf8f) file and use [`nwb`](https://www.npmjs.com/package/nwjs-builder) to package this as a `.app`. 

[image]

ambrosechua, 2016-09-03:
I don't have my SPRacingF3 EVO with me now, but the serialport listing works so I guess it works.

ambrosechua, 2016-09-03:
Oh, it also needs some modifications to [run the exit cleanup](https://github.com/cleanflight/cleanflight-configurator/blob/development/eventPage.js). Other than that, everything seems to work normally (WebGL, serial, UI, etc.)

Austinthemighty, 2016-09-03:
nice go ahed and put it in the repo, you should have permission to add changes

tcmichals, 2016-09-03:
On Linux-64-bit:
download nwjs,  untar 
cd nwjs-sdk-v0.17.0-linux-x64
export PATH=$PATH:$PWD
cd cleanflight-configurator
cleanflight-configurator$ nw ./

tcmichals, 2016-09-03:
@ambrosechua  what does nwb (nwjs-builder) do?  create a package? (yes, a newbie question)

ambrosechua, 2016-09-03:
Packs it into a distributable form. So you don't have to download NW.js then the app.

Linjieqiang, 2016-09-03:
@ambrosechua How to auto update software.

Linjieqiang, 2016-09-03:
@Austinthemighty Why not has Android suppport?

ambrosechua, 2016-09-03:
There seems to be autoupdate mechanisms for NW.js, but I think writing one from scratch would be better. A built-in autoupdater is going upstream NW.js sometime I believe nwjs/nw.js#233

ambrosechua, 2016-09-03:
@Linjieqiang Android should be a separate app and codebase written in native Java.

Linjieqiang, 2016-09-03:
@ambrosechua any plan to code the app?

Linjieqiang, 2016-09-03:
@ambrosechua Any plan for the Android app ?

Austinthemighty, 2016-09-03:
@Linjieqiang there are plans for that, but all the code has to be rewritten.

biank88, 2016-09-15:
What about Python Kivy?
I use this framework at work with Linux (also raspberry), Windows and Android. (it supports IOS too...but I never tested it on apple products).

ambrosechua, 2016-09-15:
@blank88 Oh cool! The packaging system is also simple. Kivy seems to have good cross-platform compatibility. 

I think this would be a better option. It's also smaller than using NW.js/Electron.

Austinthemighty, 2016-09-16:
@biank88 i will look into it, would you like to join our slack channel

biank88, 2016-09-18:
Yes @Austinthemighty I'd be happy to join the channel.

Austinthemighty, 2016-09-18:
ok @biank88 what is an email you would like me to send the invite to?

rodge402, 2016-10-30:
hELLO ALL, tHIS MAYBE A DUMB QUESTION BUT i WAS TRYING TO UPDATE EVERYTHING ON MY BUILD AND WHEN i TYPED IN GITHUB CLEANFLIGHT UPDATES i SEE THERE IS A COUPLE DIFFERENT THINGS WITH UPDATED INFORMATION. i ALWAYS USE THE LATEST FIRMWARE UPDATE WHICH IS THE PRE RELEASE 1.14.0 rc2 BUT i ALSO FOUND SOMETHING ON THE CLEANFLIGHT CONFIGURATOR 1.2.3 tHIS IS WHAT IS CONFUSING ME. cAN ANYONE TELL ME WHAT THE 1.2.3 IS AND HOW DO i UPDATE THE CONFIGURATOR WITH IT?

Austinthemighty, 2016-11-04:
@rodge402 If you are having problems please create a new issue, you will get more help this way, also please don't type in all caps, it makes it hard for people to read who have eye problems.

closed 2017-02-25 by hydra