cleanflight/cleanflight-configurator #423: Special Colors are broken in the LED Tab

skaman82, 2016-10-05:
Special colors are not configurable anymore (greyed out) in the LED tab. Newest configurator (dev branch) and CF 1.13. Tested on spracing F3. Default settings, just enabled the LEDs. Also got some other bug reports about this.

@gaelj do you mind to take a look at this again?

[image]
[image]
[image]

skaman82, 2016-10-14:
@gaelj I have just seen your comment on other led issue that you have some time to work on it this weekend. Could you please also have a look on this one? I am not sure how this got broken.

gaelj, 2016-10-15:
It got broken when we changed some MSP ID's to maintain compatibility with betaflight.
Should be fixed now.

closed 2017-02-25 by hydra