cleanflight/cleanflight-configurator #424: No USB in the drop down menu

drone-mechanik, 2016-10-13:
I have the latest version Cleanflight (1.2.3) and Chrome (54.0.2840.59 - 64bit) installed on my iMac running the latest Mac OS X  Sierra (10.12). I have installed the CP210x USB to UART Bridge VCP Drivers. I also have tried to force the serial connection using a command line in the terminal. None of the above helped... And I am using a proper data USB cable and not a charging cable.

I've googled around a bit but can't find a solution to my problem? 

If everything is working normally, does the USB port show up in the drop down menu even when there isn't a Flight Controller connected? Or there has to be something connected and powered up in order for it to show in the option? 
I have of course had my FC powered up and connected to my iMac when I was trying to get it to communicate with Cleanflight.

Please help!!! Thanks in advance for all your input!

* it's a TBS Vendetta, with the Powercube, the FC is Colibri.

codepreneur, 2016-10-20:
I am having exactly same problem. Any ideas?

[image]

codepreneur, 2016-10-20:
Actually I tried different USB cable and it works!

[image]

codepreneur, 2016-10-20:
The problem is described here: https://www.rcgroups.com/forums/showthread.php?t=2602386 and you can download required drivers here: http://www.silabs.com/products/mcu/pages/usbtouartbridgevcpdrivers.aspx

hydra, 2016-10-29:
Read the manuals and use rcgroups.com for support please, not here.

closed 2016-10-29 by hydra