cleanflight/cleanflight-configurator #426: GUI backup and restore missing?

unitware, 2016-10-22:
Hi I cannot see the backup and restore options in the configuration view anymore (branch development) is this intentional or an error by me?

jtrjr, 2016-10-25:
I'm experiencing the same issue. In the "Setup" window there is no longer a backup or restore option.

Nokistar, 2016-10-26:
Yes me to. I have the same problem.
I also cannot change the minimum trottle... I can't save the change.

jtrjr, 2016-10-27:
Hey guys, further research by reading the changelog shows that they did away with the backup and restore process in the latest version. Apparently, it wasn't working as it was intended. There also appears to be no way to go back to the previous version. I ended up rebuilding mine from scratch.

Jerry

tommyseus, 2016-10-27:
Which method is now the best to create and restore backups?

jtrjr, 2016-10-27:
Tommy,

I don't have any idea. I'm new to all of this stuff. The best I can tell, there is a way to do it from the "CLI" process but I haven't gotten that far. I've been more concerned with getting my settings tweaked so I could get my bird in the air.

Jerry

Nokistar, 2016-10-27:
Betaflight...

hydra, 2016-10-29:
backup and restore was killed so that people focus on the RIGHT way of doing backups, which is NOT via the old MSP commands but should be by the new Parameter Groups in the firmware.  They need exposing via MSP, @ledvinap did some work on it in the firmware but it's not done yet.  In the mean time there's no point keeping the old broken code.

closed 2016-10-29 by hydra

unitware, 2016-10-29:
ok, I see. But is there an open issue regarding this elsewhere? It's an important function that went missing and I'd love to see a discussion thread that synchronizes the work somewhere. The label should be regression I guess?