cleanflight/cleanflight-configurator #427: Newbee question about configurator update 1.2.3

rodge402, 2016-10-30:
HELLO ALL, I posted this in another section but i'm sure it wasn't the right post to ask this question so I figured I would try it this way. Anyway, THIS MAYBE A DUMB QUESTION BUT I WAS TRYING TO UPDATE EVERYTHING ON MY BUILD AND WHEN I TYPED IN GITHUB CLEANFLIGHT UPDATES I SEE THERE IS A COUPLE DIFFERENT THINGS WITH UPDATED INFORMATION. I ALWAYS USE THE LATEST FIRMWARE UPDATE WHICH IS THE PRE RELEASE 1.14.0 rc2 BUT i ALSO FOUND SOMETHING ON THE CLEANFLIGHT CONFIGURATOR 1.2.3 tHIS IS WHAT IS CONFUSING ME. cAN ANYONE TELL ME WHAT THE 1.2.3 IS AND HOW DO I UPDATE THE CONFIGURATOR WITH IT?

Austinthemighty, 2016-11-27:
OMG what is with the caps, the 1.2.3 is the configurator and the 1.14.0 is the soft ware that goes on the drones Flight controler. hope this helps

rodge402, 2016-11-27:
Thanks all for the input

closed 2016-11-27 by rodge402