cleanflight/cleanflight-configurator #428: Can't enable Ublox protocol in Configuration tab

Granluke, 2016-10-31:
Hey out there, 
I wanted to install my GPS Ublox 7m mini, but when I enabled the Uart Port and the GPS in the config tab I wanted to change to the Ublox protocol, but when I save and restart it just goes back to NMEA. Anyone a solution for this? 
I just ordered a FTDI USB adapter to check the Module over u-center, but isn't there any other solution. The green led on the gps module is on and when I set Baudrate to 9600, I see that the GPS in Cleanflight goes green (on the top where Baro Mag etc ). When setting other baud rates the GPS symbol on the top goes out. Thank you in advance!
[image]

closed 2017-02-25 by hydra

eduardoscamargo, 2017-05-11:
I am having the same problem in 2.0.3. What was the solution, @Granluke? In my case the GPS icon also remains gray.

mikeller, 2017-05-11:
@eduardoscamargo: Depending on what flight controller you are using, it is most likely the case of GPS support having been removed for a lot of targets, since they don't have enough flash space for the improved flight code _and_ advanced features like GPS.