cleanflight/cleanflight-configurator #436: Save and Reboot button missing on Windows Tablet

pilotgary, 2016-12-09:
I have a Chuwi 8" Hi8Pro Tablet for use at the flying field.  When I try to configure my board I can connect and make changes, but there is no "Save and Reboot" button at the bottom of the screen to make the changes permanent.  I have tried the tablet in desktop and tablet mode and with Chrome windowed and maximized.  Windows 10 on the tablet is up to date as is Chrome.

Is there an option to have the save and reboot button moved into the header instead of the footer?

Austinthemighty, 2016-12-12:
can you provide a screen shot

pilotgary, 2016-12-13:
I've got the tablet with me today, but not a flight controller.  I'll get some screen shots to you as soon as I can.

pilotgary, 2016-12-14:
Here are some screen grab.  The problem is occurring when Chrome is maximised. When I turned maximise off the window was bigger than the screen with the save and reboot button partially visible in landscape, but I couldn't resize the window to see it all as the handle was off screen.
When I reduced the screen resolution the window resized and I could see the whole button.  Unfortunately, when I returned the screen resolution to native the window the was off the screen again and maximising the app lost the save an reboot button.

All screen grabs are fullscreen.

[image]
[image]
[image]
[image]

Austinthemighty, 2017-01-03:
that is odd i will look into i, what is your screens resolution

skaman82, 2017-01-03:
Just tested this on my 8" Windows tablet just now and the interface works as intended. Sure, the portrait orientation is broken, but it is impossible to fit everything into that tight space - it is not a mobile app after all.
[image]

pilotgary, 2017-01-06:
I think I've found the problem.  While checking the screen resolution (1280x800) I noticed in the display settings that the "... size of text, apps and other items" was set to 200%.  
[image]
I reduced this to 100% and the "Save and Reboot" button appeared with the app maximised".  I increased the scale back to 150% and then 175% and the button remained.  
[image]
Taking it back to 200% caused it to vanish again.
Perhaps skaman82 could test if this behaviour is common to his tablet or intermittent across models.
Although small, 175% is a useable size, especially using a pointer rather than a finger. 
With this workaround, I'm happy for you to close the ticket or mark it as low priority.
Thank you for your time and great software.

closed 2017-02-25 by hydra