cleanflight/cleanflight-configurator #447: Would it be possible to redefine the RC curve?

nateimig18, 2017-03-15:
As in one that focuses on taking parameters like "center-stick" slope; 'max-stick' end points; & "center-linearity" width? I've been making matlab scripts for both Kiss & Betaflight that find the range of curves based on these first two parameters, and noticed theres (respectively) room for improvement.

I developed a curve that always keeps these first two parameters constant while varying this "center-linearity" width on much broader range. I'll attach the derivation document and a link to the project with the scripts:
https://github.com/nateimig18/AcroRcCurveFinder

[rcCurve_Redefine.pdf](https://github.com/cleanflight/cleanflight-configurator/files/846090/rcCurve_Redefine.pdf)

hydra, 2017-04-01:
yes, this one looks like one for the betaflight guys, they love this stuff.  I'm in agreement that some user-friendly way of changing the curves would be nice.  not sure who has time at the moment though.

hydra, 2017-04-29:
closing due to lack of interest.

closed 2017-04-29 by hydra