cleanflight/cleanflight-configurator #453: Wrong extension in BlackBox saved files

McGiverGim, 2017-04-04:
The extension for saved files is BFL. I think that it must be CFL (or at least TXT, like it was in version 1.x).

The code is here:
https://github.com/cleanflight/cleanflight-configurator/blob/master/tabs/onboard_logging.js#L410-L412
```
    function prepare_file(onComplete) {
        var suffix = 'BFL';
        var prefix = 'BLACKBOX_LOG';
```
I don't know if you want that I make a PR for this or simply you will change it in other commit.

Regards!

mikeller, 2017-04-05:
Good point. It was changed away from text to stop it wrongly being 'recognised' as 'text file' in certain OS, causing a text editor to be launched (and probably crash due to the binary nature and size of the file) if it was accidentally double - clicked.

My suggestion would be to use a non-firmware specific extension, and aim to keep the format compatible between the different firmwares. How about `.BBL`? @hydra, your thoughts?

McGiverGim, 2017-04-05:
What's the idea? To have only one blackbox app or two?

mikeller, 2017-04-05:
Keeping the format the same would mean having only one app - or at least two apps based off an identical code base, with different branding.

McGiverGim, 2017-04-10:
@hydra did you seen this issue? Do you prefer an extension different for CleanFlight or unify both with a .BBL extension? I can make the PR and continue the discussion there...

McGiverGim, 2017-04-11:
PR created, we can continue the discussion there ;)

hydra, 2017-04-29:
done

closed 2017-04-29 by hydra