cleanflight/cleanflight-configurator #454: Cleanflight Freezes

GJH105775, 2017-04-08:
Some important details that might make this issue worth ignoring:
I'm using Debian Linux, and Chromium web browser.

Problem:
The app freezes when I click the failsafe tab.     After clicking this ta I can't click any of the others while the tab tries to load for an indefinite time.    To use the app I am able to disconnect from the board and reconnect.     This happens on multiple cleanflight boards so it is likely the app.    

If you need more information feel free to message me here, or if I don't respond try GJH105775@gmail.com

GJH105775, 2017-04-08:
Just checked and this problem is also on WIndows 10 running google chrome 64bit and cleanflight configurator version 1.2.4

hydra, 2017-04-29:
probably out of date firmware or firmware/configurator version mis-match - re-open or comment if it's still an issue.

closed 2017-04-29 by hydra