cleanflight/cleanflight-configurator #457: Configurator 1.2.4

taileron, 2017-04-13:
Currently only 2.0.0 and 1.2.3 are available for download.
CF 1.4.2 needs configurator 1.2.4 - this version is completely missing in github at present ....

McGiverGim, 2017-04-13:
The latest branch of 1.x version is here https://github.com/cleanflight/cleanflight-configurator/tree/master_v1.x

taileron, 2017-04-13:
great .... 1000 thanx

shocky111, 2017-04-14:
wow, what a mess...been a mess for everything between 1.3.2 up to the fixed release of 1..4.2, and now this...combined by hateful  team cleanflight responses on facebook around the first of the year. Looks like I will be moving away from cleanflight sooner rather than later.

hydra, 2017-04-15:
I've tagged the v1.2.4 release, for some reason this wasn't done at the time of the release.

closed 2017-04-15 by hydra