cleanflight/cleanflight-configurator #460: Reboot without reboot sequence fail

MauroMombelli, 2017-04-15:
I am working on a seriuslypro evo3 and i uploaded my own firmware. Then i wanted to re upload the letest version of CF, so i jumped the boot pin, connected over UART1 with a FTDI chip (DFU is not seen by the system), selected the upload without reboot sequence, loaded the firmware and the upload failed.

Then i downloaded the firmware and manually fleshed with stm32flash without issue, and the board now works normally.

Maybe the system does not support USART upload?

hydra, 2017-04-15:
works fine, probably you forgot to set the manual baud rate of 256000 for the F3.

closed 2017-04-15 by hydra

MauroMombelli, 2017-04-15:
you are right, i was not setting manual baudrate, used only no reboot sequence. Why does not work is i don't set the manual baudrate? isn't this a bit counterintuitive?

hydra, 2017-04-18:
communication with the FC to put the device into bootloader via the running firmware happens at one baud rate, communication wih the FC to flash it using the CPU built-in bootloer happens at a different rate.

MauroMombelli, 2017-04-18:
But I'm using "no reset" and biit pin shorted; this eliminate the first
communication that send the R. Also the baudrate used for flashing is
dynamically found by the flash software on the first character received.