cleanflight/cleanflight-configurator #464: Lost battery monitoring

gjhaynes, 2017-04-18:
Battery monitoring no longer works on NAZE F1 since CF 2.0.0 upgrade

McGiverGim, 2017-04-18:
What version of NAZE do you have? I have a NAZE32 REV6 10DOF and it works for me. Can you explain better your issue?

gjhaynes, 2017-04-18:
Skyline32 Advanced.
Hmm, I just re-flashed the firmware and now it appears to be monitoring battery voltage. Strange!
Though the scaling options are all greyed out on the Power & Battery page. Is this normal in 2.0.0? i.e. do I need to configure scale values in the CLI now?

McGiverGim, 2017-04-18:
I observed the same issue: all options grayed. I don't know if they are aware of this.  I changed it by CLI.

gjhaynes, 2017-04-18:
Yep, I did the same. Had to disable current monitoring in CLI as well. Rather odd that the options to enable/disable vbat/current monitoring have been removed from the GUI config page.

bigd942, 2017-04-24:
Have the same issue where the battery setting are not showing in the tab in the configurator. I flashed a Lux Race V2 yesterday without issue and the battery settings showed as desired. Flashed a Furious Radiance (PICOBLX) today and they aren't showing, same Chromebook used for both. Disabled through CLI as above no issue. Screen dump attached.
[image]

hydra, 2017-04-29:
the battery settings are present, but there is no code yet to allow you to change them.  this will be done in due course.  use the cli for now.  we had to release as-is due to time constraints.

Jynxed1, 2017-09-01:
Unable to calibrate voltage on SP Racing F3 Delux, numbers are frozen at 100% cant change it.

FishButt, 2018-07-20:
I've got a quad with skyline32, crashed it half a year ago, it sat untouched while I was waiting for new parts and some free time. Plugged it in, Cleanflight shows 0 voltage when I connect my battery, if I run status shows same stuff Voltage: 0 * 0.1V (0S battery - NOT PRESENT), even though I can spin motors from Motors tab (unrelated but Motor Arming icon doesn't seem to be turning on when I arm motors/enable motor control, don't remember did it lit up before or not, but pretty sure it did). Receiver doesn't get any power as well, I had everything set up for PPM and it was working before. My battery was connected to ESC's and they were connected to motors and controller, from controller to receiver, everything was getting power from one battery, but now if I plug in battery, no USB, ESC's turn on, green led on all of them, and they beep once every couple of seconds, awaiting signal, but controller doesn't get any power. I've tried flashing different firmware's but actually with all versions of NAZE firmware starting from 2.0 and up magnetometer and barometer aren't available, don't think that I really need barometer, but still changing firmware didn't help, tried going back to 2.0.3 that I had before but still the same result, I can spin motors, with battery connected from motors tab, but if I disconnect USB controller doesn't get any power, and only gyro and accelerometer are enabled... 
So it's definitely not a wiring problem, nor battery or ESC's fault, since I can spin motors from tab... I don't remember ever configuring power & battery stuff, it seemed to work with default settings before, but now it seems that skyline32 doesn't register when battery is connected, I can also connect another battery to receiver and when USB is plugged in I get all signals in CleanFlight from my transmitter, so it's not a communication problem either ... don't think that it's firmware fault, tried downloading old Cleanflight and flashing my board with older firmware but it's still not registering battery... Also tried connecting 5V battery straight to controller, didn't do anything, it seems it now only gets powered through USB only
Am I missing some changes in Cleanflight or did my skyline32 decide to die suddenly?)

closed 2020-05-24 by stale[bot]