cleanflight/cleanflight-configurator #465: GPS map not working with latests versions of Chrome

McGiverGim, 2017-04-19:
The latest versions of chrome do not allow to load an external resource into a sandbox page. The map.html page included into the gps.html page is a sandbox as seen in the manifest.

For more info, see the warning at top of: https://developer.chrome.com/apps/manifest/sandbox

There're three things forbidden: 
- The load of the `.js` with the map api
- The inline script code into the HTML (can be solved using a `.js` file)
- The `iframe` with the map (can be solved changing it by a `webview`). 

The two last problems are easy to solve, but I couldn't solve the first. Maybe someone more experienced could.

Garogat, 2017-06-05:
I fixed it using a webview instead of the sandbox mode. To load the map I simply used the partition permissions and loaded the files directly from the app content.
I already added the request. #485

McGiverGim, 2017-06-05:
I tried a similar solution than you without luck. I will try your code to learn. 

Thanks!

McGiverGim, 2017-07-14:
This was solved. I close the issue. Thanks again!

closed 2017-07-14 by McGiverGim